rjwats/esp8266-react #2: Network Scanner doesn't work on esp32

KipK, 2019-02-10:
Hi,

Seems there's an issue with Network Scanner on esp32. It times out on browser side " Device did not return network list in timely manner"

Different errors logs from esp:
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
[E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9
[E][ESPmDNS.cpp:100] enableArduino(): Failed adding Arduino service

For information, had compile error in Windows because of Time.h from external lib being replaced by original time.h at build time. To fix it, I had to delete manually Time.h in the piolib Time folder.
There's also a warning that can be fixed by this : https://github.com/me-no-dev/ESPAsyncWebServer/pull/438/commits/6ad3cd6b5773e1354c09d4071e592d3e8638f169
Still not merged since time.

Good work so far, I really like the way you did it.  
Have you thought porting it to esp-idf framework ?

KipK, 2019-02-10:
I got more information. It doesn't work when it's not connected with wifi like first boot. 
If I configure the network after saving it still don't.
It gives this in the log at startup:
```
Starting software access point
Starting captive portal on 192.168.4.1
E (5246) event: mismatch or invalid event, id=63
E (5247) event: default event handler failed!

```
But if I reboot it after that, then it works and the log above doesn't appear anymore.

rjwats, 2019-02-12:
Hi,

Yes, I probably should have documented that you need to delete ".piolibdeps/Time/Time.h" to get this building under windows. 

I've been trying to modify platformio.ini to get it to ignore Time.h (using src_filter build option) but with no success so far. Stopping short of forking PaulStoffregen's time library to delete Time.h (which some [people seem to have done](https://github.com/xoseperez/espurna/issues/6)) - I'll add it to the documentation for now. 

Thanks for the info about the WiFi boot issue too, will keep an eye on that one!

rjwats, 2019-04-10:
espressif/esp-idf#2916 has been resolved. 

Updating to the latest IDF should prevent the errors from being shown.

closed 2019-04-10 by rjwats