rjwats/esp8266-react #3: OTA update

uncodead, 2019-03-01:
Hello,

What do i need to do to enable OTA update? 

My page settings:
enabled: true
port: 8266
pass: esp-react

My platformio.ini:
upload_flags = 
  --port=8266 
  --auth=esp-react
upload_port = 192.168.0.6
build_flags=
  #-D NO_GLOBAL_ARDUINOOTA
  -D ENABLE_CORS

I get error:
Error[1]: Begin Failed

rjwats, 2019-03-01:
Your settings look OK, I'd check that your esp module had enough flash (1mb variants don't have enough for OTA and error with that message) and also consider the possibility of your computers firewall blocking the connection.

uncodead, 2019-03-01:
Yes, i have avaliable memory. I have checked when the esp8266 starts this
lines appears at Serial monitor:

Serial.println("Starting OTA Update Service");

But this one doesnt:

Serial.println("Starting");

I think that OTA service didnt start correctly

rjwats, 2019-03-01:
The line:

Serial.println("Starting");

Is in inside the on start callback, you should not expect to see this on startup - rather when the OTA process successfully begins.

What platform are you on (Windows (version) / Linux)? Have you tried disabling your firewall temporarily to rule that out?

rjwats, 2019-03-01:
Love the brew project BTW, I had a similar project in mind...  I'm using one of these to control my fermentation chamber at the moment:

https://github.com/rjwats/esp8266-react/tree/fermentation-chamber

uncodead, 2019-03-02:
Nice! I'm doing a brew controller. I'll control a heat element with a ssr,
weather pump with a relay, and persistent a brew receipe.

I'm using your framework to save time.

After weekend I'll try a OTA update without firewall.

Thank you so much

uncodead, 2019-03-02:
Have you already tried BrewPiLess?

rjwats, 2019-03-03:
I'd seen BrewPi before, that looks like a neat port to the ESP chip!

rjwats, 2019-04-10:
I'm assuming this is resolved, please reopen if you still need help.

closed 2019-04-10 by rjwats