rjwats/esp8266-react #5: Readme documentation states incorrect ENDPOINT_ROOT path for local hosting

atsteen, 2019-04-09:
The readme lists editing Endpoints.js in order to run a local instance of the front end against a remote back end.  The edit, as written, results in an unexpected response (default html response).  In my case I had to include a "rest" dir append to make things work.  Can someone confirm this is the case?  

_Existing directions in readme.md_
The endpoint root path can be found in Endpoint.js (./interface/src/constants/). This needs to be the root URL of the device running the back end, for example "**http://192.168.0.6**".

_Suggested edit to readme.md_
The endpoint root path can be found in Endpoint.js (./interface/src/constants/). This needs to be the root URL of the device running the back end, for example "**http://192.168.0.6/rest/**".

rjwats, 2019-04-10:
Thanks Andrew,

In fact editing ".env.development" is the correct approach now. I adjusted this and forgot to update the readme.

I have updated the readme with new instructions, can you review this I'll close the ticket down.
 
Rick

atsteen, 2019-04-10:
Tested, working great, issue resolved :)

closed 2019-04-10 by atsteen