rjwats/esp8266-react #7: Strange behavior with SPIFFS

uncodead, 2019-04-10:
I have noticed strange behavior when saving configuration files (json) several times in the loop() method - eg every second. The file simply corrupts and the esp8266 stops responding.

In some cases the httpserver stops serving static files such as html, javascript, etc.

Did you face it?

rjwats, 2019-04-10:
I would be wary about saving to the same sectors of flash memory that frequently in your code. All NAND flash has a limited durability and will fail permanently after enough writes:

https://en.wikipedia.org/wiki/Flash_memory

Out of interest, what is it you are trying to write so frequently?

uncodead, 2019-04-10:
I have a status file that i need to persiste the atual state of my brew, in case of eletrical fail

rjwats, 2019-04-10:
Some people have experienced similar issues with ESP32:

https://github.com/espressif/arduino-esp32/issues/1781
https://github.com/nodemcu/nodemcu-firmware/issues/915

There might be some suggestions in those threads which could be of use.

atsteen, 2019-04-10:
FWIW, I second the "avoid frequent writes" mantra.  Unless I'm mistaken, the library used in this project to interface with spiffs is the [Arduino ESP8266 Core library](https://github.com/esp8266/Arduino).  This library appears to enable [write-caching](https://github.com/pellepl/spiffs/wiki/Performance-and-Optimizing#read-and-write-cache) which could solve both the timing and wear-leveling issues discussed.  Good luck, I'd be interested to hear how things end up!

uncodead, 2019-04-15:
The problem is the simuntaneous save in SPIFFS. I changed the loop logic and worked! Thank you.

rjwats, 2019-04-15:
Glad you figured it out!

closed 2019-04-15 by rjwats