rjwats/esp8266-react #13: 'NTPStatus' is not a class, namespace, or enumeration

waltercassiano, 2019-04-24:
Hi 

I got this error when I tried build.

atsteen, 2019-04-24:
Hello Walter, what IDE are you using?  VSCode, Atom, other?  Perhaps share the verbose build log output with us?  Thanks.

waltercassiano, 2019-04-24:
Hello @atsteen 

I'm using VSCode. 
I change on platformio.ini the Lib NtpClient to https://github.com/gmag11/NtpClient#2.5.1 and it works.

So, I'll share later.
 
Thanks

rjwats, 2019-04-24:
Looks like the NtpClient is being updated to work with non-blocking APIs, however it has an additional dependency which is not yet documented and is released at the moment as 3.0.0 beta. I'm going to update the dependencies in platformio.ini to lock to versions which should resolve this.

rjwats, 2019-04-24:
Latest commit should fix the build issues - have tested fresh builds for ESP32 and ESP8266 under VSCode.

Would you mind testing if this works for you?

waltercassiano, 2019-04-26:
Hi 

Nice, it works

tks

rjwats, 2019-04-27:
Thanks Walter, I'm closing the issue.

closed 2019-04-27 by rjwats