rjwats/esp8266-react #14: Upgrade to latest NTPClient lib once stable 3.x.x release is available.

rjwats, 2019-04-24:
The latest release of NTPClient uses non blocking APIs and we should upgrade to this once a stable release is available for testing.

beta-tank, 2019-06-14:
Hi, thanks for your project. I am new here and I’d like to know about the benefits of using the `NTPClient` library.
I use the default implementation from `sntp.h` and there are no problems.
Initialization can be done like this:
```
 configTime(7 * 3600, 0, "time.google.com");
```
or you can manually call all init functions:
```
sntp_stop();
sntp_setserver(0, "time.google.com");
sntp_set_timezone(7);
sntp_init();
```

Then you should wait for the time initialization to happen. I make it in the `setup` function in a synchronous way (because my functionality needs the actual time), but it can be made in the `loop` in an async way with a callback.
Here is my implementation, fairly straightforward:
```
time_t getTime(){
  #ifdef ESP8266
    return sntp_get_current_timestamp();
  #endif
  #ifdef ESP32
    time_t now = 0;
    return time(&now);
  #endif
}
int last = 0;
  while(getTime() < 1500000000){
    if((millis () - last) > 1000){
        last = millis ();
        Serial.println(getTime());
        Serial.println("Waiting for sntp.");
    }
    yield();
  }

rjwats, 2019-06-14:
Gmag's NTP client implementation provides a refresh interval and
retry-on-failure behaviour on top of the network level NTP handler.

beta-tank, 2019-06-15:
Thanks for the explanation. I'll try to use NTPClient library in my project to simplify the migration to your react framework.

proddy, 2019-12-29:
SNTP is now native in the ESP8266 and ESP32 arduino core. See https://github.com/rjwats/esp8266-react/issues/47#issuecomment-568173582. After the PR has been merged you can close this

rjwats, 2020-01-20:
Won't do - sntp support now implemented

closed 2020-01-20 by rjwats