rjwats/esp8266-react #16: Migrating to ArduinoJson6

uncodead, 2019-04-27:
Hi, i'm planning to migrate my project to framework's new version, but i've some classes with attributes like this:

JsonObject *_mashSettings;

How can i migrate in this case (Arduinojson6)? See exemples:

https://github.com/uncodead/biabrewex/blob/master/src/MashService.cpp
https://github.com/uncodead/biabrewex/blob/master/src/MashService.h

the screen of this code:
[image]

Whats the best way to migrate this code? Can you help-me?

rjwats, 2019-04-27:
See the following diff: https://github.com/rjwats/esp8266-react/commit/e68b7627f2751f5366ab3870709a0bdffe0ad7de

The diff of SettingsPersistence.h demonstrates most of the changes required to work with the new ArduinoJson v6 APIs.

There is an upgrade guide on the ArduinoJson which explains how to use the new MemberProxy objects:

https://arduinojson.org/v6/doc/upgrade/

Let me know if you need any further info!

uncodead, 2019-04-27:
Thanks, mission acomplished.

But, i have some problem when i run interface:

[image]

uncodead, 2019-04-27:
Resolved upgrating npm/node. Thanks

closed 2019-04-27 by uncodead

rjwats, 2019-04-27:
Thanks, glad you managed to fix the issue.