rjwats/esp8266-react #18: Save settings inside arduino

uncodead, 2019-04-27:
Hi,

I have a settings file:
https://github.com/uncodead/biabrewex/blob/master/src/BrewUNO/BrewSettingsService.cpp

.. and i want to update this settings inside of Arduino, not from user interface (web app). Whats is the best way to do this?

eg: _brewSettingsService->update()

thanks

rjwats, 2019-04-27:
SettingsPersistence.h is a mixin you can use to save JSON settings to the
file system.

You can extend it, as SettingsService does, to write to the SPIFFS
filesystem.

uncodead, 2019-04-27:
do you think to implement at framework this feature?

rjwats, 2019-04-27:
If I'm understanding your question, you want to write JSON to the
filesystem within a function in your code - such as when an event occurs or
on a periodic basis.

As you have extended SettingsService, you have already extended the
SettingsPersistence.h mixin which will have provided you with writeToFS and
readFromFS functions to serialize the JSON config to the file system.

Just call writeToFS() within a public function in BrewSettingsService.h to
expose it....

public:

void saveStuff() {
  writeToFS();
}

then call

_brewSettingsService-> saveStuff()

If I have misunderstood what you are trying to do, give me some more
details.

uncodead, 2019-04-27:
All right! thank you.

closed 2019-04-27 by uncodead

aurban6, 2019-04-28:
Hi
I use SPIFFSEditor.
`#include <SPIFFSEditor.h>
void setup()
{
.......
server.addHandler(new SPIFFSEditor(SPIFFS, "admin", "admin");
.......
}`

uncodead, 2019-04-28:
I put the method writeToFs as a public in src/SettingsPersistence.h