rjwats/esp8266-react #26: More elaborated dashboard

uncodead, 2019-05-21:
Hi, I would like to create a more elaborate dashboard for my application, with some gadget like the image below. My question is: with your experience with the framework, what is the best approach to elaborate a dashboard like this maintaining all the characteristics of the framework?

[image]

rjwats, 2019-05-21:
This is an exercise for the reader really as it will depend on your use case :)

I have had decent success with the react wrapper for jschart2 for charts:

https://github.com/jerairrest/react-chartjs-2

I've used this on my [fermenator project](https://github.com/rjwats/esp8266-react/tree/fermentation-chamber) with good results. 

I'd be inclined to use [the grid](https://material-ui.com/layout/grid/) to ensure you get good fluid UI when using it on a mobile device or desktop.

Good luck!

uncodead, 2019-05-23:
Thanks! 

[image]

closed 2019-05-23 by uncodead