rjwats/esp8266-react #27: Build error on windows

uncodead, 2019-05-31:
I have a windows dev environment and when i build the framework i get this error:
[image]
So, i need to rename the time.h file and rename at this line:
[image]
Because windows is not case sensitive. What's the best way to avoid this problem?

rjwats, 2019-05-31:
See the comment in the documentation regarding this issue under windows. I've not found a better solution but am going to investigate a pre-build python script to allow for the

uncodead, 2019-05-31:
Yes, I followed this post to solve the problem. But it's not good for users to make this rack on their machine

rjwats, 2019-05-31:
Not ideal, but neither is forking or copying the time library to delete the file. Like I say - i think a pre-build script is probably the easiest solution which will resolve the issue at build time. Will get to this at some point - currently working on [adding security](https://github.com/rjwats/esp8266-react/pull/29).

rjwats, 2019-06-02:
I've developed a solution which should resolve this issue at build time, it involves a pre-build script which deletes the offending Time.h file before the build begins:

[timelib_fix.py](https://github.com/rjwats/esp8266-react/blob/master/timelib_fix.py)

Looks like this works on fresh builds on my windows box, let me know how this works for you.

uncodead, 2019-06-03:
It works, thanks

closed 2019-06-03 by uncodead