rjwats/esp8266-react #34: Using framework as a library

beta-tank, 2019-06-16:
Hi, is there any way to use this framework as a connectable library?
I think this way it might be more convenient for users, because you can update it easily and initialize in your code only the parts that you need (thanks for the examples).
Now if I start creating a project on top of this framework, I won't be able to update its sources in the future or won't be able to use a separate git repository (because I have to use this repo to be able to update the framework sources).
I think it can be easily done for the ESP code, but I don't know how to do this for the React part (I don't have much experience with this framework).
What do you think about this idea?

rjwats, 2019-06-16:
Yes, this is definitely not a library that you can just import and go. Nor is it light weight :) but I enjoy React because of the libraries and tools available.

You could do what I do and fork the repo, keeping your projects in branches pulling in the changes when needed:

https://help.github.com/en/articles/configuring-a-remote-for-a-fork

I've pondered whether to wrap this up as a library, I'm sure I could write some tooling to serve the react content in a static manner (compiling it into the binary) but that would mean the UI wouldn't be customizable by importee... tricky one.

beta-tank, 2019-06-17:
I think the React part can be implemented as a collection of js modules without app.js (it can be provided as an example). Then the user can create their own implementation with only the required components. You can specify the folder for the user scripts and static content, and then your frontend build scripts will continue working.

rjwats, 2019-07-06:
Related to: https://github.com/rjwats/esp8266-react/issues/42

FernandoGarcia, 2022-03-14:
Hi!

Please take a look in this [framework](https://github.com/maakbaas/esp8266-iot-framework). It's very similar to your project and can be used as library. Maybe it can be useful reference for you.

Best regards.

FernandoGarcia, 2022-03-15:
I did a [PR](https://github.com/rjwats/esp8266-react/pull/281) for it. Please take a look.

rjwats, 2022-03-18:
Thanks for this. I agree that there would be value had in wrapping this up as a PIO (and also Arduino) library. There's enough in the framework for that to be worthwhile, but as you've pointed out there's plenty out there which does similar, so I've never felt particually compelled.

A few thoughts:

- It feels like the UI should really be pre-built and released with the source code package (WWWData.h) for this to be feasible. In other words, don't require the user who is including the library to have NPM and to build the React UI - they arn't going to be changing it.
- The dependency and its sub-dependencies would be very large (especially for esp8266s) so ideally I'd want to provide more than just a WiFi, OTA Updates and NTP for it to be worthwhile. In particuar I had thought that a declarative API for defining your project-specific configuration. Somthing akin to: https://github.com/ayushsharma82/ESP-DASH but for configuration rather than dashboards.

It feels like any "includable library" should either be a built artefact, or a downstream project. The latter giving the flexiblity to add value as described above.

FernandoGarcia, 2022-03-18:
Hi!

`"It feels like the UI should really be pre-built and released with the source code package (WWWData.h) for this to be feasible. In other words, don't require the user who is including the library to have NPM and to build the React UI - they arn't going to be changing it."`

For now could it be solved using [build flags](https://github.com/maakbaas/esp8266-iot-framework/blob/master/docs/installation-guide.md#build-options) as on framework linked above?

The script will look for a "rebuild interface flag" and if it is not present will skip all build process so `npm` will not be necessary.

In this case the system will use a given fully featured `WWWData.h` file and data folder.

Best regards.