rjwats/esp8266-react #36: AP mode not working

uncodead, 2019-06-20:
My ESP is configured to always start AP mode. On startup i have this logs:

Starting software access point
Starting captive portal on 192.168.4.1
WiFi connection dropped, stopping NTP.

But i cant find ssid with my cell phone.

rjwats, 2019-06-21:
Hi Bruno,

Your project is looking really good!

Are you having this problem with the mater branch? I made a series of recent fixes, tweaks and upgrades. One of them addressed an issue around the access point. The problem was due to WiFi channel switching when the ESP tries to reconnect, which it does continuously.

The aforementioned fix is here: 

https://github.com/rjwats/esp8266-react/blob/master/src/WiFiSettingsService.cpp

Let me know if this helps.

uncodead, 2019-06-25:
Thank you @rjwats I put your name on the credits!

Unfortunately I could not match my version with your master. I'll have to program myself to get the master version... :/

rjwats, 2019-06-25:
Setting this repo as an upstream origin might help you with your merge process:

https://help.github.com/en/articles/configuring-a-remote-for-a-fork

uncodead, 2019-06-25:
Do you know this problem:

[image]

rjwats, 2019-06-25:
Thanks for pointing that out Bruno. It looks like esp32's arduino core implements the stop function, esp8266 handles this in the destructor instead. Resolved in master with an if-def.

rjwats, 2019-06-25:
Did you make any progress merging the AP fix into your project?

If it's of interest the new implementation of WiFiSettingsService allows you to assign an empty SSID in [wifiSettings.json](https://github.com/rjwats/esp8266-react/blob/master/data/config/wifiSettings.json) which will effectively disable the WiFi station mode. This allows the Access Point to run totally uninterrupted. The fallback (when unable to connect) is to control the reconnect frequency which is done in the class and controlled by WIFI_RECONNECTION_DELAY in [WiFiSettingsService.h](https://github.com/rjwats/esp8266-react/blob/master/src/WiFiSettingsService.h) it should be fairly easy to backport these changes into your application. Let me know if you need any more info.

uncodead, 2019-06-26:
Hi rjwats, im working on merge your master with my project version (hard work).

I have an other problem in your master:

[image]

rjwats, 2019-06-26:
I'll take another look at master this evening, though it seemed to be behaving fine last night. In the mean time, you might try a clean rebuild by running:

'pio run -t clean'

I removed what I assumed was a redundant reference to the esp's WiFi library from the main class which may have cause this new issue, will investigate later.

uncodead, 2019-06-26:
yes, i got a new environment (platformio and vscode) to build your master and got this erro.

I think that erro is related to #ifndef ESP8266 i dont kknow where in workspace it was configured as true. To by pass this error i need to comment this lines. I dont know why...

rjwats, 2019-06-26:
Hi Bruno,

Odd, I have just pulled a fresh copy from master and it's builds fine under Windows for me.

The first error, INADDR_ANY not resolving, shouldn't be an issue as as WiFiSettingsService.h pulls in IPAddress.h from the Arduino core package which provides this.

The "ESP8266" define is configured by platformio based on your platform, in this situation it is behaving correctly. You are building on the esp8266 core, which provides 'strcpy_P(dest, src)' as part of string.h In this case the function is not provided by the DateStrings library because the Arduino core provides it already.

You should see what I mean if you find the following file on your file system (assuming it is there!):

'C:\Users\Rick\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include\sys\string.h'

If you are getting this issue on your other projects too I suggest your packages are corrupt in some way and need re-installing.

rjwats, 2019-06-26:
For reference I am on Espressif 8266 v2.2.2 and my packages are in my user folder:

C:\Users\Rick\.platformio\packages
and 
C:\Users\Rick\.platformio\platforms

So re-installing platformio may not cause these to be re-fetched. If you delete them platformio will refetch them on build (note: delete, not rename!)

uncodead, 2019-06-27:
> For reference I am on Espressif 8266 v2.2.2 and my packages are in my user folder:
>
> `C:\Users\Rick\.platformio\packages`
>
> and
>
> `C:\Users\Rick\.platformio\platforms`
>
> So re-installing platformio may [...]

I was with old ESP8266 platform version... i got the latest and solve the issue.

Now im working on update my arduino code and i'll use my version of interface...

Mabe you can consider enable/disable the JWT auth (my project doesnt needs)

Thanks!

rjwats, 2019-06-27:
Yes, that's a reasonable feature request. That should be opened as a separate issue, there's 3 things getting discussed here.

rjwats, 2019-06-27:
Hi Bruno,

I'm closing this issue and opening another one to track the feature request:

https://github.com/rjwats/esp8266-react/issues/39

Please re-open if you are still having the AP issues.

closed 2019-06-27 by rjwats

uncodead, 2019-06-27:
Thank you rj!