rjwats/esp8266-react #37: 'strcpy_P' was not declared in this scope

uncodead, 2019-06-25:
When trying to compile for the first time I get the error below:

[image]

have you face this problem?

rjwats, 2019-06-25:
I believe the only errors i have seen with the time library were due to [this issue](https://github.com/rjwats/esp8266-react/issues/27) with the time library, which was resolved in the master branch recently.

I will try a fresh build this evening to see if I get the error. I take it you are building under Windows?

uncodead, 2019-06-25:
I uninstalled platformio and refreshed my workspace and the error disappear. Thanks

closed 2019-06-25 by uncodead

fkeller234, 2019-06-28:
I'm having the same problem, I made the reinstallation and it didn't solve the problem!
Did you find a solution?