rjwats/esp8266-react #41: Delay to connect with wifi

uncodead, 2019-07-05:
Hello!

After updating my project with your master, I realized that esp is taking a few minutes to connect to wifi. Did you notice that too?

rjwats, 2019-07-06:
There is a deliberate delay if there is a failure to connect to the WiFi. See the code in WiFiSettingsService.cpp to see how this works. The delay is fixed to 60 seconds.

The delay is required to prevent the device continually channel switching when in auto-reconnect mode - this causes the access point stability issues. 

There might be an improved solution to this with a [back-off retry approach](https://en.wikipedia.org/wiki/Exponential_backoff) which would allow for more frequent retries without crippling the access point.

I'll look at implementing this soon.

uncodead, 2019-07-06:
Is there any problem set ap provision mode to never? When I setted to
'never' my wifi connected instantly.

rjwats, 2019-07-06:
Setting the access point provision mode to "never" will prevent the access point from coming up in the event of WiFi connection failure. Disable it if you like, but you won't be able to re-configure the device if you change your WiFi password (without re-flashing via serial).

The back-off retry approach might give you better luck. I've not had any issues with connecting to WiFi first time personally so this issue I suspect it's related to WiFi channel switching. I'm planning on implementing the exponential back-off retry approach, which might help in your situation.

closed 2019-07-09 by uncodead