rjwats/esp8266-react #42: Extensible interface

uncodead, 2019-07-05:
Have you thought about leaving the interface extensible so that I do not need to change the .js files in the framework? Example: you create a .js file so that I include my routes, a special folder for me to extend the new pages of my project.

Today I always have to merge when I upgrade to your master

rjwats, 2019-07-06:
Yes, this has come up, this issue is related to:

https://github.com/rjwats/esp8266-react/issues/34

I think your suggestion of having a convention for the user to follow is a good one. It would be more friendly to have well defined extension points for users who aren't practiced at merges.

I will draft something...

rjwats, 2019-07-06:
**Work In Progress**

Implementing a "demo project" in [this branch](https://github.com/rjwats/esp8266-react/tree/ft_demo_project), which should go some way to resolving:

https://github.com/rjwats/esp8266-react/issues/34
https://github.com/rjwats/esp8266-react/issues/42
https://github.com/rjwats/esp8266-react/issues/31

Check out the branch and comment if you would like. The idea is a demo project, isolated as much as possible from the main interface code, with its own routing and side menu:

https://github.com/rjwats/esp8266-react/tree/ft_demo_project

It won't prevent you from ever having to do work when pulling from master. I'm planning on keeping the framework up to date with changes to React and MaterialUI which will indicate you'll need to update your code too - no avoiding that! However it will encourage a better coding practice (isolation) to reduce the number of complex manual merges you will need to do in future.

[image]

rjwats, 2019-07-07:
In addition to working on the interface, I'm exploring the idea of moving the bulk of the C++ code into the lib directory. To allow users to isolate their project's C code from the framework's:

[image]

It's a work in progress, but it is promising I think.

uncodead, 2019-07-08:
> **New Feature - Work In Progress**
> 
> Implementing a "demo project" in [this branch](https://github.com/rjwats/esp8266-react/tree/ft_demo_project), which should go some way to resolving:
> 
> #34
> #42
> #31
> 
> Check out the branch and comment if you would like. The idea is a demo project, isolated as much as possible from the main interface code, with its own routing and side menu:
> 
> https://github.com/rjwats/esp8266-react/tree/ft_demo_project
> 
> It won't prevent you from ever having to do work when pulling from master. I'm planning on keeping the framework up to date with changes to React and MaterialUI which will indicate you'll need to update your code too - no avoiding that! However it will encourage a better coding practice (isolation) to reduce the number of complex manual merges you will need to do in future.
> 
> [image]

do you think about enable/disable jwt authentication?

rjwats, 2019-07-08:
There is another open issue for that.

rjwats, 2019-10-06:
The recent changes pushed to master have resolved this issue. 

Any more ideas on making this framework easier to extend are welcome!

closed 2019-10-06 by rjwats