rjwats/esp8266-react #46: MQTT implementation

carlosatta, 2019-08-02:
Hi all,

I'm trying to implement MQTT with Arduino MQTT library (https://github.com/256dpi/arduino-mqtt).

I'm not a C/C++ developer, and I'm stuck with the status page (https://github.com/carlosatta/esp8266-react).

For now I've created the settings and status page and MQTT settings service, but seam impossibile to me read the mqtt connection status inside the MQTTStatus.

Some one can help me please?

rjwats, 2019-08-02:
Hi Carlo,

I'm not seeing any of your back end code committed to your '[mqtt](https://github.com/carlosatta/esp8266-react/tree/mqtt)' branch can you push it up so I can take a look?

What is the error you are seeing?

Rick

carlosatta, 2019-08-03:
Hi,

sorry my fault. Now I've push all the files.

In the file https://github.com/carlosatta/esp8266-react/blob/mqtt/src/MQTTStatus.cpp i wold like to read the mqtt connection status but seams to be always false.

In the file https://github.com/carlosatta/esp8266-react/blob/mqtt/src/MQTTSettingsService.cpp  I start the mqtt connection and seams to be all ok and i can change configuration in FE at /mqtt/settings url.

rjwats, 2019-08-03:
One possible problem is how you are storing your MQTTClient instances in your objects. These are being passed in as pointers but being dereferenced and values assigned to new variables. Try converting these to pointers.

carlosatta, 2019-08-03:
Done!

If u are interessed in this feature I can clean up my code and make a pullrequest.

Thanks.

rjwats, 2019-08-04:
Getting MQTT integrated could be useful yes! Your project looks interesting too!

If you are wondering why this was broken, it was because you effectively make 3 different copies of the client instance when [using variables](https://www.tutorialspoint.com/cprogramming/c_variables.htm). C isn't like garbage collected languages where you are almost always passing around references - variables are areas of memory.

I want to rework the code so some features can be enabled as build time options. I have a good idea about how to achieve this - there is another issue open for disabling the authentication feature I added recently - a valid feature request.

Feel free to open a PR, it looks like decent work! I might pull it into a branch for later integration once I've addressed making features optional.

Feel free to close this issue down if it's all sorted now :)

rjwats, 2020-02-24:
Is it worth trying to support SSL with MQTT?

In my experimenting with [async-mqtt-client](https://github.com/marvinroger/async-mqtt-client) I have managed to get SSL "working" but it consumes 20k of RAM on the esp8266 which is rather impractical.

In addition it looks like [AsyncTCP](https://github.com/me-no-dev/AsyncTCP) doesn't yet have SSL support, meaning the MQTT client can't support it under ESP32: https://github.com/marvinroger/async-mqtt-client/issues/164

Is MQTT without SSL a credible offering?

proddy, 2020-02-24:
It does require a lot of additional CPU and RAM which as you said is precious on an ESP8266. In the end I removed it from my implementation as it couldn't get it stable together with my old web interface. There are however implementations that have done it ([espurna](https://github.com/xoseperez/espurna/wiki/MQTT#ssl-support), [tasmota](https://github.com/arendst/Tasmota/wiki/Securing-your-IoT-from-hacking)) with a lot of hacking and code.

I would say drop it.

rjwats, 2020-04-30:
PR is open which introduces MQTT and WebSockets including an example project which demonstrates the new features:

**Feature Branch:** https://github.com/rjwats/esp8266-react/tree/ft-mqtt-websockets

**PR:** https://github.com/rjwats/esp8266-react/pull/102

I'd be very interested in feedback before I push this change to master as it's a fairly significant diversion for the core of the framework. The change makes the framework more event orientated and factors out the serializer concept from the core classes so data can be serialized in a different ways for different use-cases (I found home assistant usually wanted a different format to the REST endpoints, especially with auto-discovery - it's quite prescriptive).

I think it represents a positive change and provides better internal decoupling (SOLID), but I would love some comments if anyone who's had experience with the framework has time to have a mess around with it.

lorol, 2020-04-30:
Hi @rjwats just a note you may find interesting.
I used an optimized ESPAsyncWebServer fork, based on: https://github.com/sascha432/ESPAsyncWebServer 
see the results, first one is the original lib.
[image]
[image]

lorol, 2020-04-30:
I need to learn a lot until be able to use your great framework, maybe first I need to buy a new computer, because the initial build takes 20-30 min on the one I have. 
Also I need to fully understand how to modify/create react UI js, for me it looks just hacking. Somewhat I am missing the advantage of this huge JS stuff buzzwods around. Feel a bit blind - to add a button or menu ... wow is not easy, still source code and building. 

But It works. I will play with it, Thanks a lot for WS.

rjwats, 2020-05-01:
Wow, that's quite an impressive memory improvement over the original.

Lots of moving strings to flash I see, I really should consider this for
the framework too. I know @Proddy was looking at this before.

lorol, 2020-05-01:
Hi @rjwats, any practical MQTT testing, guide especially for the mentioned "Home Assistant's auto discovery feature"?
I succeed to to connect the device to free broker (mqtt.eclipse.org), also can run one at home network.
What next? I am bit lost ... and have to go far down.

rjwats, 2020-05-01:
Hey lorol,

Home Assistant uses a pair of topics, one to configure the device (the
"set" topic) and a topic for the device to publish it's state (the "state"
topic). You could test it manually by poking data into the MQTT topics
using an MQTT client if you didnt' want to install Home Assistant itself.

Here's the docs for the "light" device:

https://www.home-assistant.io/integrations/light.mqtt/

A light (the demo implements this), publishes it's state to the state topic
with a JSON message which looks like this:

homeassistant/light/a9e350cc/state:

{"state": "off"}

The payload for the "set" topic is the same but is used to control the
device, so you can turn the light on and off by publishing to the state
topic with the same payload:

homeassistant/light/a9e350cc/set:

{"state": "on"}

It's awkward to test this without Home Assistant, but you could use a
client such as MQTTExplorer to play around with it if you wanted :)

Rick

proddy, 2020-05-01:
@rjwats I'll try out the latest PRs and provide some feedback. I fully support the effort you put into decoupling and serializing the interfaces. Can't beat clean code and it's good source for anyone coming from hobby Arduino C and wanting to learn more on modern OO design patterns and C++11/17. I'm planning to swap out my current web UI with ESP8266-React soon. 

MQTT is a wolf is sheep's clothing. I found when I implemented an mqtt service on some on my own projects I ended up writing a library with a lot of code to handle mqtt publish queues, retries, QOS 1/2 acknowledgements etc.

@lorol good find on the memory optimized asyncwebserver fork. True, I'm a bit obsessed when it comes to squeezing out every byte of heap from the ESP8266s and using flash memory where ever possible. I'm actually fed up with the ESP8266s (they are 6 years old now!) and can't wait for those cheap ESP-S2s to hit production.

rjwats, 2020-05-17:
MQTT support added to the framework along with example code.

closed 2020-05-17 by rjwats