rjwats/esp8266-react #47: NTP Timezone

carlosatta, 2019-08-03:
Hi all,

I think a good feature is the possibity to set timezone in setting service.

I will try to help with this ASAP.

rjwats, 2019-08-03:
NTP does not deal with timezone, it uses UTC so this is probably a front-end feature. 

The UI may be simply improved by using moment.local() when formatting dates in the front end. This would probably be fine for most use cases as users would see dates in their appropriate timezones.

carlosatta, 2019-08-03:
yhea, this is right,

but seems NTP have a method to set time zone "NTP.setTimeZone(+2);"

In my current project I need to watch the midnight, so the set of the local time is usefull for me in the BE/ESP code.

Maybe someone else need this feature too.  what do u think?

rjwats, 2019-08-03:
Possibly, dates and time are always more complex than they seem :) For simple cases - I generally like stick to UTC at the back end converting to local dates for display.

Given your use case you will need to translate from UTC to your local timezone to detect time of day. That comes with all the fun of dealing with daylight savings in locales that have them!

carlosatta, 2019-08-03:
I think this issue can be set as closed.

rjwats, 2019-08-04:
Using moment.local() in constants/TimeFormat.js to get a localized time in the UI (with little effort) might be worthwhile:

Something like this should do the trick:

`moment.unix(unixTime).local().format(TIME_AND_DATE);`

Supporting fully localized time in the back end might be worthwhile - it's a bigger change.

Up to you if you want to close it. I'll make a note to sort out localized time rendering at some point.

proddy, 2019-11-25:
@carlosatta I needed real times in my ESP code (not the web) for logging to MQTT and Serial so added Timezone and daylight saving time support to my [NTP](https://github.com/proddy/EMS-ESP/blob/dev/src/Ntp.cpp) library and had a parameter to define the timezone. But to be honest if I would implement this again I would stick to UTC and let the browser handle the rendering. In the browser I used this flunky code:

```js
 var d = new Date(0);
 var c = new Date();
 var timestamp = Math.floor((c.getTime() / 1000) + ((c.getTimezoneOffset() * 60) * -1));
 d.setUTCSeconds(timestamp);
```

didn't know about `moment.local()` !

proddy, 2019-12-21:
FYI I added timezone support in my fork (which I'll open up soon). Also removed the NTPClientLib and replaced with a smaller class as the ESP Arduino core natively supports SNTP since 2.4.x with LWIP2.

rjwats, 2019-12-21:
I see that the NTP library has been deprecated in favour of the SDK's
provided utils.

It would be good to get this updated with a PR!

proddy, 2019-12-29:
Still need to make sure the ESP32 port is compatible. Then the PR.

rjwats, 2020-01-19:
It looks as if the callback and status functions added to esp-idf (for ESP32) have not made their way into arduino-esp32 yet:

https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/system_time.html

- sntp_set_time_sync_notification_cb
- sntp_get_sync_status
- sntp_set_sync_interval

These would be of particular interest, as they would enable the existing functionality to be maintained. Displaying the status, showing the last sync time, controlling the synchronisation interval with config.
 
Without these, especially the callback and the ability to set an interval the only thing we can universally support is the core feature of NTP synchronisation with a fixed timeout - but of course with the TZ functionality added.

There's SNTP_UPDATE_DELAY in lwip which you can override with an extern'ed function reference but I've had little success under ESP32 - works fine with 8266, the Arduino HAL provides a named function as you've seen: "sntp_update_delay_MS_rfc_not_less_than_15000"

I'm in half a mind to add the time zone feature and ditch the status features so I can get this working in a uniform way across both esp32 and esp8266. The status feature seems less important.

proddy, 2020-01-19:
+1 for ditching status features.

IMHO the extra functionality is probably not that critical. As engineers we tend to look at the problem from a technical angle and add things because we can, whilst from an end-user perspective the feature and UX requirements may look different and often less complex. I see this at work every day.

For example NTP, the user just wants to have real times perhaps to show in on screen messages or logging. So now() should always give back the adjusted time. They would care less about the refresh interval or the status as long as it works. If I look at Ubiquiti's expensive networking hardware and software (like my USG router I own) the only thing you can adjust is the NTP server's addresses. So I agree you probably don't need a NTPStatus service and web panel.

Generally speaking there is more simplification we can do. Take for example the System Status. The CPU speed, chip size, SDK version, whether its an ESP8266 or ESP32 etc is also probably not the useful to most people. As an end-user I would expect they would be more interesting in the overall health of the system and see things in a dashboard like:
 - System Load (as a %)
 - Wifi Strength (as a %)
 - Available memory (as a %) and colour coded to red/orange/green 
 - #failed login attempts
 - which modules are active (ntp, ota, mqtt...)

rjwats, 2020-01-20:
This has been implemented.

closed 2020-01-20 by rjwats