rjwats/esp8266-react #55: Convert String into char[] *

uncodead, 2019-11-18:
In order to make the project more stable and reliable, ESP8266/Arduino programming recommends avoiding String objects to prevent a memory leak. Is there any idea to replace String objects to char *?

Thank you!

proddy, 2019-11-19:
these days **std::string** is easier. See this [example](https://roboticsbackend.com/arduino-stl-library/).

uncodead, 2019-11-20:
The std:string memory consume is similar to char []?

proddy, 2019-11-20:
all stl/std libraries take some space, but its not too much and a lot simpler than writing your own string functions. If you use String correctly you won't get into problems with stability and I think rjwats knows what he's doing. However if memory is a problem then you shouldn't really use react and look at preact which has a much smaller footprint. I'm seeing if I can port some of this to preact as an experiment.

uncodead, 2019-11-20:
Are we talking about same memory? Im talking about heap memory.

rjwats, 2019-11-20:
If you are having memory leak issues I'd be interested in seeing the code and conditions under which you are experiencing the issues. 

I have a SoapUI project which I use locally to load-test the projects endpoints (1000s of requests to each endpoint) before I merge changes - memory leaks would be fairly awful. 

I haven't experienced any memory leaks to date, though it is of course possible a chance in the core or libraries used could have introduced a problem. Do you have any more details of why you think you have a memory issue? I could send you my SoapUI project (or check it in) so you can run the tests yourself.

Let me know!

proddy, 2019-11-21:
> Are we talking about same memory? Im talking about heap memory.

std::string supports “Small String Optimization” (SSO) meaning it stores short strings locally and only uses the heap for long strings. By reducing the number of small objects in the heap, the SSO reduces the fragmentation which can lead to memory problems you see with the Arduino String class.

If you want to monitor memory, put something like

in setup()

   ` static uint32_t _heap = ESP.getFreeHeap();`

in loop() 

```
    uint32_t total_memory = _heap;
    uint32_t free_memory  = ESP.getFreeHeap();

    Serial.printf("Free Heap: %d bytes initially | %d bytes used (%2u%%) | %d bytes free (%2u%%)",
            total_memory,
            total_memory - free_memory,
            100 * (total_memory - free_memory) / total_memory,
            free_memory,
            100 * free_memory / total_memory);
```

proddy, 2019-12-17:
@rjwats I'm progressing on implementing some of those changes I talked about before (centralizing settings, mqtt, syslog, telnet console etc) and will upload soon to my GitHub fork. I was wondering if you would mind sharing the SoapUI project so I can also bash the end-points to make sure its stable?

rjwats, 2019-12-17:
Looking forward to seeing what you've done and to see what can be brought over to the core. Did you find a clean approach for managing settings in the end? Doing that in a modular way looked like it would be the most difficult bit.

Hoping to find some time over the xmas break to work on some of this myself - making the build modular for a start.

I'll try and remember to share the file when I get the chance, though might not be till tomorrow as I'm busy this evening.

proddy, 2020-07-02:
Think you can close this.

closed 2020-07-02 by rjwats