rjwats/esp8266-react #57: Extending esp8266-react with Features

proddy, 2019-11-25:
First of all thank you very much Rick for building and sharing this great project.

For my home automation projects I had built something similar based on bootstrap and plain html/javascript which is now getting too big (500kb gzipped) and in need of a refactor.  I had started porting the code to use preact and milligram to cut down on the web size,  until I found this gem.

So I’m planning to extend esp8266-react with some of the features I already have in my framework whilst keeping it generic and lightweight. I’m thinking of services like MQTT, Telnet, Logging, Update, Backup/Restore, Timezone etc and making them optional.

Question: Are you still actively working on esp8266-react? Would you like to work together on a few of these or just review PRs you think would be useful? I'm fine either way.

Here's a list of things I was planning to add:

New Services:
* MQTT service (working off carlosatta’s implementation)
* Telnet Service, for replicating Serial when Serial can’t be use. Both reading commands and writing debug statements.
* Logging Service (messages to Serial, Telnet if open or SysLog). This would need to be core to all services.
* Update Service that shows the latest firmware from GitHub Releases page and download button
* Backup/Restore service that compacts JSON files into one file

Improvements:
* Switch to native ArduinoJson6 support and use message-pack serialization for SPIFFS files to save on space
* Create mock back-end server to serve JSON responses for each endpoint, so I can test without having an ESP running
* Add system stability check and boot sequence. If reset button pressed during startup it goes into factory restore
* Replace String with std::string, just because its cool
* Investigate if there will be performance gains if using PROGMEM instead of SPIFFS for Web content
* Move some of the constant strings to flash memory
* Add Reboot option somewhere in the UI
* Extend NTP to support timezone & Daylight Saving Times
* Extend WiFi with MDNS to allow <hostname>.local discovery
* Extend system screen to show % RAM and % SPIFFS left, also load average
* Wifi and MQTT automatic timeouts and reconnect
* Nice-to-have: enable/disable Features at build time
* Nice-to-have: look at reducing Web code if using preactJS and a smaller CSS template like Milligram

carlosatta, 2019-11-25:
Hi, 

I have already managed some of this task.
In the next days I should share my work with you.

proddy, 2019-11-25:
nice @carlosatta . Perhaps we could create a new branch here and collaborate together?

carlosatta, 2019-11-25:
sounds good for me!

rjwats, 2019-11-27:
Hey Proddy,

Yes, still actively maintaining. Though a busy home and work life consumes much of my time!

Some of the features you mentioned I've already considered implementing, such as "factory reset" and "soft reset". I've also considered using TypeScript for the front end, but so far have steered clear of that as i thought it might put some people off using the framework for their projects.

I think there's probably a line somewhere between what should be considered core vs application specific feature. The latter 
being features (such as MQTT) which not every project would require. Case in point; I recently added user authentication assuming it's a feature everyone would want, but was met with requests to be able to disable the feature at build time (still in backlog).

With regards to non core features. It might be worth thinking about how to modularise the application, perhaps with a feature availability service, so features can be enabled/disabled at build time.

Thoughts?

rjwats, 2019-11-27:
I like the idea of a remote logging service :)

proddy, 2019-11-27:
@rjwats agree that some services should be optional. The way [espurna](https://github.com/xoseperez/espurna) does this is via `#DEFINES` and having all the source files enclosed in `#ifdef`'s.  It's a bit clunky and makes the code hard to read but it does the job.

I do have very specific needs, that's true, for example all my projects use MQTT to communicate to Home Assistant and some of my designs can't use the Serial port which I was why I ended up using Telnet to show real-time information and debug messages to help troubleshoot. I'll start with implementing those both on esp8266-react and see how far I get.

rjwats, 2019-11-27:
Indeed... I've wanted to avoid cramming features into the framework in an effort to keep it simple and clean for everyone. However features like MQTT are examples of common usage patterns - it would be a shame for everyone who wants to use MQTT in their project to have to re-do the effort of integrating a library and the associated configuration service and UI components.

I'll see if I can come up with a good strategy for features to self-register in the framework, so they can be excluded with a build flags as you alluded to earlier. Excluding features (such as configuration pages for optional features) in the UI can be done dynamically by interrogating a feature availability service when the app first mounts.

I'll start by making authentication an optional component, then there'll be a good approach for integrating other optional features without imposing them on all users :)

proddy, 2019-12-01:
@rjwats would you be ok if I created separate GitHub issues for some of the features I listed above. Each can used as a reference, discussed and closed if you don't think it adds any value. But it will create more spam. Perhaps use [labels](https://github.com/proddy/EMS-ESP/issues) to help categorize between wish-list enhancements and bugs?

I'm working on add telnet and logging now.

rjwats, 2019-12-01:
Go for it.

proddy, 2019-12-29:
Most of this I have implemented, in https://github.com/proddy/MyESP2. I'll continue to add separate issues for the individual features.

closed 2019-12-29 by proddy