rjwats/esp8266-react #62: AP is open on initial install/upload

proddy, 2019-11-30:
Thanks for the recent changes. One thing I did notice is on a fresh install (i.e. pio run -t erase, build & upload FS, upload firmware) I don't get the captive login. It creates an Access Point with the default EMS8266 SSID like ESP_XXXXX with no password. I haven't looked into the code why this happening yet.

rjwats, 2019-11-30:
Odd! I had never erased one of these esp8266 chips until you mentioned this. Same effect for me!

Flashing my other chips (without erasing) gives normal behaviour. It must be something to do with the build or upload in PlatformIO. Maybe there's a flag missing to initialise the flash properly. I'm not too sure yet.

I managed to get my erased chip working though PlatformIO after flashing a random sketch to it through the Arduino IDE using the settings below:

[image]

It again breaks after erasing... I've done [some reading](https://docs.platformio.org/en/latest/platforms/espressif8266.html) and nothing has leapt out at me yet. I've tried explicitly telling the linker which [.ld script](https://github.com/esp8266/Arduino/blob/master/tools/sdk/ld/eagle.flash.4m.ld) to use to no effect. Will keep looking and see if I can find an answer.

I guess the work-around for now is to not erase the flash :)

proddy, 2019-11-30:
Not sure its flash related. The SPIFFS FS and json files are loaded correctly. I think the problem is in the wifi.getmode() which is always returning WIFI_AP even before the WiFi has been initialized. I'll deep dive tomorrow.

rjwats, 2019-12-01:
Indeed, I noted the config was loading fine also. Uploading a different sketch changed the persistent opmode resulting in the device starting up subsequently in station mode - so the APSettingsService code was then initialising correctly on startup.

The espressif SDK stores it's persistent WiFi config in a separate area on the flash (/* wifi   @0x405FD000 (12KB) */ ) and it would appear that when erased with 0xFFs this value becomes (or defaults to) a 3 (WIFI_AP_STA) - at least that's what this code was demonstrating on my device:

`
void setup() {
  Serial.begin(115200);
  WiFi.begin();
}
void loop() {
  Serial.println(WiFi.getMode());
  delay(1000);
}
`

In esp8266-react we disable WiFi settings persistence and rely purely on configuration stored as JSON in SPIFFS. However the management code does expect the device to be coming up with an opcode of 0.

I've tested and pushed a fix for this which makes sure the device always starts up in opmode 0 (by configuring the persistent default).

Thanks for pointing this one out, I never erased my devices so I never saw it!

proddy, 2019-12-01:
great that you found the root cause. I would also have expected that when all the flash sectors were erased the wifi mode would not default to STA on boot up.

proddy, 2019-12-01:
Closing this as the fix works marvelously.

closed 2019-12-01 by proddy