rjwats/esp8266-react #69: Use PROGMEM for storing web files

proddy, 2019-12-17:
One thing I would really like some help implementing is moving the web files from SPIFFS to the code, and storing this in Flash memory. The advantages I believe are
- no need to build and upload SPIFFS separately
- uploading new web changes to the ESP Filesystem will overwrite existing config files
- possibly speed

To do this it would mean creating header files for the following web endpoints containing the compressed data:
 app/manifest.json
 app/icon.png
 css/roboto.css
 fonts/* all 3 files
 js/* - merging the 3 js files into one

I did something similar before using [gulp](https://github.com/proddy/EMS-ESP/blob/master/tools/webfilesbuilder/gulpfile.js) but I'm sure there are more elegant ways these days using webpack. I'm just not an expert in that area.

rjwats, 2019-12-18:
Yes, it's a good idea to have this option. Any "speed" benefits are probably tenuous - though the other points are valid. In addition, this feature would help with the "Project as a library" request which really is the same goal as you have - being able to patch the device with new code/UI, leaving the config as-is. I would be looking to implement this for all UI artefacts rather than a pre-defined list (so it's extensible for those adding features to the framework). Route registration would have to be code-generated too - all quite doable though.

I did notice the build artefact for the ESP32 was rather large - perhaps prohibitively so to make this work with default flash partitioning - will have to take a look at that too.

rjwats, 2019-12-22:
I've made good progress on this BTW. Have developed a WebPack plugin which
extracts the assets and static resources, gzips and hex-encodes as PROGMEM
to header files in to a specified location relative to the npm build path.
Benefit of this approach is that it's extensible, you can add any
additional static resources or assets and the'll be added to the build.

Final bit to do is the code generation - for creating the request handlers.
Hope to get around to this tonight.

Still a little concerned there won't be enough space on ESP32 devices
without changing the partitioning but I'll cross that bridge later - there
may be a reason the build is so large

proddy, 2019-12-22:
excellent! I'm struggling with time too but did manage to implement the Console (telnet/serial) last night. I'll open the repo up when it's all working properly. To be honest I did get a little carried away and not everything is worthy of PRs as it breaks your rule of being simple & flexible.

I was thinking the same about using too much PROGMEM. In other ESP8266 projects I have used less partition for the FS e.g. in platformio with the build flag `-Wl,-Teagle.flash.4m1m.ld` which allocates 1019 KB sketch, 1000 KB SPIFFS. 4KB EEPROM, 4KB RFCAL, 12KB WIFI stack and 2052 KB OTA & buffer.

rjwats, 2019-12-26:
Integrated this into the build process with a separate pre build script. The platform io build now triggers the npm build and PROGMEM generation (if enabled). I've made it an build option which can be enabled with "-D PROGMEM_WWW".

Especially on the ESP32, the artefact size is going to cause a problem. Moment is a very large dependency which doesn't support tree-shaking so it's an obvious one to ditch - not really using it for much so it can probably go. Using system provided fonts instead of Roboto would reduce the artefact size by 48k so thats probably worth doing too.

### After removing moment:

**ESP8266**

From SPIFFS:
PROGRAM: [=====     ]  51.0% (used 532948 bytes from 1044464 bytes)

From Progmem:
PROGRAM: [========= ]  85.4% (used 891480 bytes from 1044464 bytes)

**ESP32**

From SPIFFS:
PROGRAM: [========  ]  84.8% (used 1111112 bytes from 1310720 bytes)

From Progmem:
PROGRAM: [==========]  100.0% (used 1311244 bytes from 1310720 bytes)

### After removing moment & Roboto:

**ESP8266**

From SPIFFS:
PROGRAM: [=====     ]  51.0% (used 532948 bytes from 1044464 bytes)

From Progmem:
PROGRAM: [========  ]  80.9% (used 844456 bytes from 1044464 bytes)

**ESP32**

From SPIFFS:
PROGRAM: [========  ]  84.8% (used 1111112 bytes from 1310720 bytes)

From Progmem:
PROGRAM: [==========]  96.4% (used 1264076 bytes from 1310720 bytes)

Still need to decide exactly if/how to cut the npm (or cpp?) artefact size down, 96.4% is much too close for comfort on the ESP32. I'd love to know why it's so large, even without the web artefacts it's 84%(!!!). Will investigate.

One disadvantage of using PROGMEM for storage of this data, which I hadn't considered until testing this this evening, is that we'll have to reserve enough free space for an OTA update who's binary now includes bundled WWW data (as opposed to a separate OTA operation for the SPIFFS upload).

[image]

rjwats, 2019-12-29:
Hi @proddy 

Can I get your feedback on the PR I've just opened. 

It's a little messy on the plugin side. I'm still trying to get the travis build to work now that the build script also builds the user interface using npm.

The PR also includes some changes inspired from your fork:

- some improvements to the platformio.ini file
- defaults for the users which were missing previously

Your changes have made me very aware of some of the basic things I can do to improve the framework - such as moving strings to flash! The current in-lining of strings is really wasteful. I guess my Java background translates to some bad habits in C-land!

It's really cool being able to blank the device and flash it once and see an operational interface - no SPIFFS upload step!

proddy, 2019-12-29:
sure, will look this afternoon

proddy, 2019-12-29:
had a play with the ft-progmem-www. That's some very nifty scripting, nicely done! There are definite advantages to this approach. True the OTA will be larger but you probably only need 1kb for SPIFFS now leaving much more buffer space over for the OTA image. It'll be interesting to see if the response time serving the pages is any faster as there is less file-server seeking. I'll continue exploring and report back any feedback here.

thanks again!

rjwats, 2019-12-29:
One thing I noticed which I have just solved is that when running  "pio run
--target erase" it was running the build interface step.

I guess we really want it to run only when the build/upload targets are
run. So I've modified the script to only run the npm step on those targets.

rjwats, 2019-12-29:
I've merged this after testing it as much as I could. 

Please re-open if you find any issues which require fixing :)

closed 2019-12-29 by rjwats

proddy, 2019-12-30:
if the web gets too large you could try playing with preact using https://github.com/preactjs/preact-compat . I've seen examples where the bundle is reduced by a third.

rjwats, 2019-12-30:
I saw you mention that before, possibly worth looking at at some point :)

I'm going to look at "optional features"  next, though probably don't have
any time today.