rjwats/esp8266-react #73: Upgrade interface to TypeScript

rjwats, 2019-12-29:
Upgrading the interface to TypeScript might be a nice addition. 

There is inconsistent use of PropTypes in the interface at the moment and TypeScript is much better way of type constraining things.

I'm interested in whether people think this is a good idea. I'm a TypeScript fan, so I'm predisposed towards making this change.

proddy, 2019-12-30:
+1 for switching to typescript. Not only does it make development easier like catching problems instead of waiting for runtime failures it will make the code much more easier to follow as its more self-describing.

rjwats, 2020-01-05:
WIP: https://github.com/rjwats/esp8266-react/tree/ft_typescript

rjwats, 2020-02-06:
This is very close now, restructured the file system for the react side along with the TypeScript conversion. 

I think it's much clearer, what do we think?

proddy, 2020-02-08:
looks very good. I like what you did with the types.ts separation. 

minor thing, I noticed one typo in the README in your settings example at the end, between `esp8266React.getWiFiSettingsService()` and `esp8266React->getWiFiSettingsService()`

rjwats, 2020-02-08:
Thanks @proddy I'm pleased with the restructuring too. Much easier to work on the UI code now, you can't go back once you've used TypeScript.

Have opened a [PR ](https://github.com/rjwats/esp8266-react/pull/89) so I can give it a looking over and run some manual tests on various configurations before merging. Comments welcome!

closed 2020-02-09 by rjwats