rjwats/esp8266-react #74: Compiler errors

justoke, 2019-12-30:
I have tried to get my [forked ](https://github.com/justoke/esp8266-react) repo working but can't get past these numerous compiler errors. I have a cloned version from a few months back which is fine. Note the interface project can be built and run just fine and it works against the ESP32 that is already running the firmware.

I found some references to Time.h versus time.h and saw that you have a py script to handle this file but not sure if this is relevant. I stopped the compiler just after the first failures started as it is even longer that the excerpt shown.

Processing esp32ota (platform: espressif32; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 
Verbose mode can be enabled via `-v, --verbose` option

warning: Ignoring missing SConscript 'timelib_fix.py'
File "C:\users\sbens\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 149, in <module>
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.11.1 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), tool-openocd-esp32 1.1000.20190708 (10.0), framework-arduinoespressif32 2.10004.191002 (1.0.4), tool-esptoolpy 1.20600.0 (2.6.0)  
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <NtpClientLib> 2.5.1
|   |-- <Time> 1.6 #c0436e6
|   |-- <WiFi> 1.0
|-- <ArduinoJson> 6.13.0
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP> 1.1.1
|   |-- <ArduinoJson> 6.13.0
|   |-- <Time> 1.6 #c0436e6
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <FS> 1.0
|-- <framework>
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <AsyncTCP> 1.1.1
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.13.0
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <AsyncTCP> 1.1.1
|   |   |-- <ArduinoJson> 6.13.0
|   |   |-- <Time> 1.6 #c0436e6
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESPmDNS> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <Update> 1.0
|   |-- <NtpClientLib> 2.5.1
|   |   |-- <Time> 1.6 #c0436e6
|   |   |-- <WiFi> 1.0
|   |-- <Time> 1.6 #c0436e6
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
```
Building in release mode
Compiling .pio\build\esp32ota\src\DemoProject.cpp.o
Compiling .pio\build\esp32ota\src\main.cpp.o
Compiling .pio\build\esp32ota\lib798\ESP Async WebServer_ID306\AsyncEventSource.cpp.o
Compiling .pio\build\esp32ota\lib798\ESP Async WebServer_ID306\AsyncWebSocket.cpp.o
In file included from c:\users\sbens\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\locale_facets_nonio.h:39:0,
                 from c:\users\sbens\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\locale:41,
                 from c:\users\sbens\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\regex:45,
                 from .pio/libdeps/esp32ota/ESP Async WebServer_ID306/src/WebHandlerImpl.h:25,
                 from .pio/libdeps/esp32ota/ESP Async WebServer_ID306/src/ESPAsyncWebServer.h:461,
                 from .pio/libdeps/esp32ota/ESP Async WebServer_ID306/src/AsyncJson.h:38,
                 from lib/framework/SettingsService.h:13,
                 from lib/framework/AdminSettingsService.h:4,
                 from src/DemoProject.h:4,
                 from src\DemoProject.cpp:1:
c:\users\sbens\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\ctime:64:11: error: '::clock' has not been declared
   using ::clock;
           ^
c:\users\sbens\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\ctime:65:11: error: '::difftime' has not been declared
   using ::difftime;
           ^
c:\users\sbens\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\ctime:66:11: error: '::mktime' has not been declared
   using ::mktime;
           ^
c:\users\sbens\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\ctime:67:11: error: '::time' has not been declared
   using ::time;
           ^
c:\users\sbens\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\ctime:68:11: error: '::asctime' has not been declared
   using ::asctime;
           ^
Compiling .pio\build\esp32ota\lib798\ESP Async WebServer_ID306\SPIFFSEditor.cpp.o
c:\users\sbens\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\ctime:69:11: error: '::ctime' has not been declared
   using ::ctime;
           ^
c:\users\sbens\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\ctime:70:11: error: '::gmtime' has not been declared
   using ::gmtime;
           ^
c:\users\sbens\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\ctime:71:11: error: '::localtime' has not been declared
   using ::localtime;
           ^
c:\users\sbens\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\ctime:72:11: error: '::strftime' has not been declared
   using ::strftime
```;

Any suggestions?

rjwats, 2019-12-30:
Hi,

Yes, those are the errors you get building under windows when "Time.h" conflicts with "time.h" and the incriminating file needs to be removed from the imported Time library. That is the purpose of the script which the error mentions.

The scripts have moved, it might be that your platformio.ini needs updating to include the new paths (they are under a "scripts/" sub directory now. Other than that i'd only suggest you close PlatformIO, delete it's system folders from your project directory and try again.

[image]

rjwats, 2019-12-30:
On a side note, there will shortly be changes to move away from the "Time" and "NtpClientLib" libraries, instead employing the SDK provided ones which are now much better. So the problematic libraries and duly require work-around will go too. Let me know if you manage to work around the issue or if you think it's a problem we need to look at.

justoke, 2019-12-30:
> On a side note, there will shortly be changes to move away from the "Time" and "NtpClientLib" libraries, instead employing the SDK provided ones which are now much better. So the problematic [...]

Thank you for your prompt response. I'm always in favour of simpler conventions, the less code there is the better.

I managed to fix the script reference and cleaned out the folders as suggested. I get this error now:

```
Verbose mode can be enabled via `-v, --verbose` option
ImportError: No module named pathlib:
  File "C:\Users\sbens\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 149:
    env.SConscript(item, exports="env")
  File "C:\Users\sbens\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\sbens\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\Users\sbens\Documents\PlatformIO\Projects\esp8266-react\scripts\build_interface.py", line 1:
    from pathlib import Path
```

File "C:\Users\sbens\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 149:
```
for item in env.GetExtraScripts("pre"):
    env.SConscript(item, exports="env")
```

The platform.ini attached is fairly standard, with just a section for my module specifics. Is there some dependency on [env] perhaps whereas I am building for [env:esp32ota] ?

[platformio.txt](https://github.com/rjwats/esp8266-react/files/4010432/platformio.txt)

rjwats, 2019-12-30:
You must be using python 2.7 - either upgrade to 3.8 which comes with it
bundled or you can pip install the pathlib library:

"pip install pathlib"

rjwats, 2019-12-30:
https://pypi.org/project/pathlib/

justoke, 2019-12-30:
Wish this was simple as it seems.

**python 2.7**
```
$ pip install pathlib
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip 
will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Requirement already satisfied: pathlib in c:\python27\lib\site-packages (1.0.1)
```

**python 3.8**
```
C:\Users\sbens\AppData\Local\Programs\Python\Python38-32\Scripts>pip3 install pathlib
Collecting pathlib
  Using cached https://files.pythonhosted.org/packages/ac/aa/9b065a76b9af472437a0059f77e8f962fe350438b927cb80184c32f075eb/pathlib-1.0.1.tar.gz
Installing collected packages: pathlib
  Running setup.py install for pathlib ... done
Successfully installed pathlib-1.0.1
WARNING: You are using pip version 19.2.3, however version 19.3.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.
```

I have multiple versions of Python installed - 2.7 and 3.8, default which seems to be 2.7 and it claims to have the pathlib package installed, but I still get the same exception each time. I've tried amending the path environment variables and restarting PIO. No luck so far.

justoke, 2019-12-30:
Resolved - removed Python 2.7 and restarted PIO.

closed 2019-12-30 by justoke

rjwats, 2019-12-30:
Nice one! I wouldn't normally have used a > 2.7 API but I think 2.7 is EOL
soon. Glad you got it sorted.

justoke, 2019-12-30:
I have some specific interest in this project and would like to discuss some things with you. If at all possible to have a zoom call at your earliest convenience, please ping me on my github email. Thank you.

rjwats, 2019-12-30:
Sure

justoke, 2020-01-03:
Are you free sometime today?

rjwats, 2020-01-03:
Did you recieve my email?

justoke, 2020-01-03:
I got the replies via email from github and not from you directly.

rjwats, 2020-01-03:
I had sent an email to sbe (at) iventris (dot) net on the 31st of December.

Maybe it ended up in your spam folder?

justoke, 2020-01-03:
Email located and replied. Skype is good.