rjwats/esp8266-react #78: Extending the framework - begin() is protected

justoke, 2020-01-13:
In the example describing the extension [here](https://github.com/rjwats/esp8266-react#extending-the-framework), it is possible to provide overrides for the SettingsService methods with the exception of begin() as it is not marked as virtual in the base/super class.

I tried to implement the reconfigure logic but the begin method is protected in Settings service and the compiler fails
```
class MQTTSettingsService : public SettingsService {
 public:
  MQTTSettingsService(AsyncWebServer* server, FS* fs) :
      SettingsService(server, fs, MQTT_SETTINGS_SERVICE_PATH, MQTT_SETTINGS_FILE) {
  }

  ~MQTTSettingsService() {
  }

 protected:
  void readFromJsonObject(JsonObject& root) override {
    _server = root["server"] | "";
    _port = root["port"] | 0;
    _user = root["user"] | "";
    _password = root["password"] | "";
  }

  void writeToJsonObject(JsonObject& root) override {
    root["server"] = _server;
    root["port"] = _port;
    root["user"] = _user;
    root["password"] = _password;
  }

  /*     void begin() {
          // make sure we call super, so the settings get read!
          SettingsService::begin();
          reconfigureTheService();
      } */

  void onConfigUpdated() override {
    reconfigureTheService();
  }

  void reconfigureTheService() {
    // do whatever is required to react to the new settings
  }

 private:
  String _server;
  int _port;
  String _user;
  String _password;
};
```
[image]

It seems that simply using a method say setup() would provide an alternative.

rjwats, 2020-01-13:
It might be because you are trying to reduce the visibility of the
function.

It's declared to have an access modifier of public in the super class so it
must be in the public in the sub class too.

WiFiSettingsService.h has an example of this in the framework.

justoke, 2020-01-13:
That was it!  Thank you.

```
class MQTTSettingsService : public SettingsService {
 public:
  MQTTSettingsService(AsyncWebServer* server, FS* fs) :
      SettingsService(server, fs, MQTT_SETTINGS_SERVICE_PATH, MQTT_SETTINGS_FILE) {
  }

  ~MQTTSettingsService() {
  }

  void begin() {
    // make sure we call super, so the settings get read!
    SettingsService::begin();
    reconfigureTheService();
  }
 protected:
  void readFromJsonObject(JsonObject& root) override {
    _server = root["server"] | "";
    _port = root["port"] | 0;
    _user = root["user"] | "";
    _password = root["password"] | "";
  }

  void writeToJsonObject(JsonObject& root) override {
    root["server"] = _server;
    root["port"] = _port;
    root["user"] = _user;
    root["password"] = _password;
  }

  void onConfigUpdated() override {
    reconfigureTheService();
  }

  void reconfigureTheService() {
    // do whatever is required to react to the new settings
  }

 private:
  String _server;
  int _port;
  String _user;
  String _password;
};
```

closed 2020-01-13 by justoke

proddy, 2020-01-13:
you may be interested in the MQTT service I created - it doesn't use the standard JSON settings file configuration but the core code may be useful for your MQTT implementation.

https://github.com/proddy/esp8266-react/tree/proddy

justoke, 2020-01-13:
> you may be interested in the MQTT service I created - it doesn't use the standard JSON settings file configuration but the core code may be useful for your MQTT implementation.
> 
> https://github.com/proddy/esp8266-react/tree/proddy

Hi - actually I was looking at your project as rjwats did point it out to me. I'm going to try use the interface for the MQTT settings. I noticed you had added some sort of a command processor. There is a lot of very nice features there. I was just trying to get going with just the config file and a basic settings service as part of porting my project over. I'll ping you if I have any questions. Thank you.

justoke, 2020-01-13:
@proddy @rjwats There is a lot of brilliant work in these two projects. Well done and thank you! Having looked at both projects, I find myself wanting to incorporate elements of both. This makes it slightly challenging to keep up with your ongoing efforts.

My use case is to deploy a whole fleet of devices - up to 300 at a time. The configuration and so on would be a one time setup at installation and commissioning phase. Thereafter, it would not be needed as visiting the site would be impossible or logistically challenging. So then, MQTT and remote control would be preferred means of making tweaks and so. 

**Remote OTA**
I've managed to incorporate an OTA update based on checking a config file hosted on the web and then downloading the latest firmware.

rjwats, 2020-01-13:
Pull-based OTA updates are a very desirable feature to have for the framework - it's a requirement many people share. I'd be very interested to look at this!

justoke, 2020-01-13:
> Pull-based OTA updates are a very desirable feature to have for the framework - it's a requirement many people share. I'd be very interested to look at this!

Sure thing. I'm giving it a thorough test at the moment. For now I have added the remote URL to the otaSettings.json file and implemented an OTAService to manage the check, download and update.

justoke, 2020-01-13:
@proddy I have tried to build you proddy branch but ran into a number of compile issues. Attached is the build log for the esp32-release environment. There are a number of compile errors. 
[esp32-release.txt](https://github.com/rjwats/esp8266-react/files/4054383/esp32-release.txt)

The esp8266-release worked first time. Please could you check - I did build the interface and so on.

rjwats, 2020-01-13:
A comprehensive framework feature might want to:

   - schedule periodic checks for automatic updates
   - allow the user to manually check for updates and apply one if available
   - (possibly) allow the user to select from a list of available releases
   to deploy one of their choosing

I'm liking this feature!

proddy, 2020-01-13:
> allow the user to manually check for updates and apply one if available

This is something on my to-do as well. I had previously implemented something similar on another one of my projects. Basically a web page that showed the latest stable and development builds (pulled from the GitHub releases) along with the release notes and the option to download and upload using OTA over HTTP. It's just some JavaScript. The builds were built automatically using a TravisCI.

[image]

proddy, 2020-01-13:
> @proddy I have tried to build you proddy branch but ran into a number of compile issues. Attached is the build log for the esp32-release environment. There are a number of compile errors.
> [esp32-release.txt](https://github.com/rjwats/esp8266-react/files/4054383/esp32-release.txt)

I haven't ported it yet to esp32. Its mentioned as well in the README

justoke, 2020-01-13:
Thanks for the explanation. I was eager to get it going and didn't read the manual! I did manage to get the code that I was after added to my fork and the MQTT settings are now in the REACT app. Working on getting the MQTT client working.

I got rid of the Freertos.h compile errors by just including #include <FreeRTOS.h> in MQTTStatus.h.