rjwats/esp8266-react #88: Live LED status on demo - suggestion

lorol, 2020-02-04:
Hi @rjwats 

Thank you for this very interesting project!
I managed to reproduce the demo of your master head on Win and esp12e. Nice.
To get a better sense of the react, is it possible to be added to the project tab UI one extra "LED" button, in best case to show with its color change the live status when device LED is on or OFF and also to override on touching (manual toggle)?
When the slider idle is at 0, i guess it would be possible flip on/off by touching.
This way we'll see the full power of the idea in 2 directions, unless I am really off-topic :)

Is this possible only by modifying these 2 files or it is way more complex?
https://github.com/rjwats/esp8266-react/blob/master/src/DemoProject.cpp
https://github.com/rjwats/esp8266-react/blob/master/interface/src/project/DemoController.js

Your help is appreciated!
LL

rjwats, 2020-02-05:
The demo could me made more comprehensive, though it was really intended as a minimal "Hello world" example with the real world application being left as an exercise for the reader.

WRT your question, you may want to modify the header too to add some more state variables to track the additional state you'll require to do what you are suggesting. Sounds feasible.

lorol, 2020-02-05:
Is there any "easy" way, other than websocket or ugly time-pooling to get data pushed from server to the browser in near real-time?

rjwats, 2020-02-05:
I have used WebSocket for this to good effect in the past. This branch
(based on a rather old version of the framework) has "SimpleSocket" wrapper
which provides WebSocket support without loads of boilerplate required:

https://github.com/rjwats/esp8266-react/tree/rgb-lightstrip/src/lightstrip

I should probably bring a WebSocket example into the demo project based at
some point - it's a common requirement.

lorol, 2020-02-05:
WebSocket example into the demo project - will be great!

lorol, 2020-02-09:
Hi @rjwats 
How do you manage to make nice web layout with components? Is there any graphical designer tool or all is "try and error" method by looking at code and seeing the live result?

rjwats, 2020-02-09:
The project uses Material UI, plenty of good examples in the documentation.

https://material-ui.com/

rjwats, 2020-05-21:
The latest framework has changes to support WebSockets and the demo project shows how to use it to keep the UI in sync with the state on the device.

closed 2020-05-21 by rjwats