rjwats/esp8266-react #90: build_interface.py, Undefined variable problem

jake-code, 2020-02-09:
Hi, I've just started to look into your react framework and am having a problem. In the build_interface.py script I'm getting an error _Undefined variable 'Import'_ relating to the line of code Import("env").
Can you please look into this and provide some guidance to solve the problem?
Thanks for your time and effort.

rjwats, 2020-02-09:
Not one I've seen, missing a dependency in your environment is the most
likely cause.

Are you on Windows or Linux? Have you installed the prerequisites?

You'll need python 3.8.

proddy, 2020-02-09:
I refreshed my fork, on osx, and it built everything first time via VSC. Will check on Win10 tomorrow. 

As a side note, Rick, it's truly amazing how much time and effort you put into this project. If I compare to the other ESP projects using the async tcp and aysnc web server libraries yours is up there as #1. We need to get the message out.

rjwats, 2020-02-09:
Thanks for the encouragement @proddy! Good to hear you could build the latest changes under OSX. Let me know if you have any issues!

jake-code, 2020-02-10:
Hi, I'm running Linux (Ubuntu 19.10 with all updates installed). Installed python is 2.7.17 and 3.7.5 with default python as python2 (/usr/bin/python links to /usr/bin/python2.7).
As far as building the project it actually completes with success, but it reports as problems Undefined variable for 'Import', 'env' and 'BUILD_TARGETS' in build_interface.py.
If I attempt to run build_interface.py from a shell prompt it aborts with error (NameError: name 'Import' is not defined).
So although I can build the project and upload to my 8266 I can't help but wonder if everything was built as it should or I am going to be running into some unknown and unexpected problem caused by this error situation.
I've been searching for anything I can on the problematic line Import('env') but nothing has come up yet. I haven't event managed to find out where Import exists/comes from, to check if it could be a dependency issue.
I'm sorry this isn't much to go on. You mention python 3.8, is Import() something new from there? 
Once again thanks for the help.

rjwats, 2020-02-10:
I wouldn't expect you to be able to run the python file manually as
PlatformIO invokes it from within it's environment.

I'll have a go building it later on a Ubuntu VM and see how I get on. In
the meantime, perhaps update your defaults and seeing if the switch to 3.7
makes it happier.

This builds on travis with python 3.8:
https://github.com/rjwats/esp8266-react/blob/master/.travis.yml no idea
what version of Linux travis uses for its python containers though!

proddy, 2020-02-10:
@jake-code you're using python 2.7 I suspect at default. Switch to python3 for platformio by following these steps:
```sh
% python --version
Python 2.7.17                                                                                                                      
% sudo apt install python3-pip
% pip3 --version                                                                                     
pip 18.1 from /usr/lib/python3/dist-packages/pip (python 3.7)
% pip3 install platformio
```

then

```sh
% git clone https://github.com/rjwats/esp8266-react.git
% cd esp8266-react
% pio run
```

and it will build esp8266-react correctly. Just tested this on Ubuntu 19.10 eaon.

rjwats, 2020-02-10:
Cheers for that proddy!

jake-code, 2020-02-13:
Sorry for the delay in responding. Although Ubuntu 19.10 uses python2 as it's default in Platformio I've been using python3 from start, so it's not a python2 issue.
Also like I said before the Platformio project does build, reporting success and I can upload and run on my esp8266, apparently without any issues. It's just that when building it does complain about that one line in the python pre-script and I wonder if it the consequences will eventually surface as some other issue that will be hard to track down later on.
But aside from that everything seems to work pretty smooth.
I've even started to integrate an async FTP server library I coded some time ago (when I started to play around with the ESPAsyncTCP library) into the main framework menu (since you also have that lib included I thought it would be a good way to learn about the framework and at the same time add the capability to upload and download files from SPIFFS).

rjwats, 2020-02-14:
Would you mind providing the output from your terminal?

jake-code, 2020-02-14:
Of course, it's pasted below these messages that are reported on the Problems tab and relative to build_interface.py:

  Undefined variable 'Import'
  Undefined variable 'env'
  Undefined variable 'BUILD_TARGETS'

--- Terminal output ---
> Executing task: platformio run <

Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
--------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Building interface with npm
npm install

> core-js@2.6.11 postinstall /root/Arduino/toLookAt/Platformio/esp8266-react/interface/node_modules/babel-runtime/node_modules/core-js
> node -e "try{require('./postinstall')}catch(e){}"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon: 
> https://opencollective.com/core-js 
> https://www.patreon.com/zloirock 

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

> core-js@3.6.4 postinstall /root/Arduino/toLookAt/Platformio/esp8266-react/interface/node_modules/core-js
> node -e "try{require('./postinstall')}catch(e){}"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon: 
> https://opencollective.com/core-js 
> https://www.patreon.com/zloirock 

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.11 (node_modules/webpack-dev-server/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.11: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.11 (node_modules/watchpack/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.11: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.11 (node_modules/jest-haste-map/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.11: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.2 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

added 248 packages from 170 contributors, removed 128 packages, updated 359 packages and audited 918768 packages in 93.713s

51 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities

npm run build

> esp8266-react@0.1.0 build /root/Arduino/toLookAt/Platformio/esp8266-react/interface
> react-app-rewired build

Creating an optimized production build...
Compiled successfully.

File sizes after gzip:

The project was built assuming it is hosted at the server root.
You can control this with the homepage field in your package.json.
For example, add this to build it for GitHub Pages:

  "homepage" : "http://myname.github.io/myapp",

The build folder is ready to be deployed.
You may serve it with a static server:

  npm install -g serve
  serve -s build

Find out more about deployment here:

  bit.ly/CRA-deploy

Copying interface to data directory
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 2.3.2 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 2.40802.191122 (4.8.2), framework-arduinoespressif8266 2.20603.191216 (2.6.3), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20800.0 (2.8.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 6.14.1
|-- <ESP Async WebServer> 1.2.3
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ArduinoJson> 6.14.1
|   |-- <ESP8266WiFi> 1.0
|-- <framework>
|   |-- <ArduinoJson> 6.14.1
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <ESPAsyncTCP> 1.2.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ArduinoJson> 6.14.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266mDNS> 1.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
Building in release mode
Compiling .pio/build/d1_mini/src/DemoProject.cpp.o
Compiling .pio/build/d1_mini/src/main.cpp.o
Compiling .pio/build/d1_mini/lib7cf/framework/ESP8266React.cpp.o
Retrieving maximum program size .pio/build/d1_mini/firmware.elf
Checking size .pio/build/d1_mini/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [=====     ]  47.3% (used 38752 bytes from 81920 bytes)
PROGRAM: [=====     ]  51.7% (used 540380 bytes from 1044464 bytes)
======================================= [SUCCESS] Took 183.95 seconds =======================================

Environment    Status    Duration
-------------  --------  ------------
d1_mini        SUCCESS   00:03:03.951
node32s        IGNORED
======================================== 1 succeeded in 00:03:03.951 ========================================

rjwats, 2020-02-14:
Ahha! You are talking about the problems tab in the UI rather than output from "pio run". 

This makes more sense now as if an error like that occurred during the build process it would prevent it from completing successfully.

The problems view (tab) is typically populated by IDE plugins which perform static analysis on the code (checking syntax, known APIs and linting to name a few) and is not related to the running of the build command. Whatever static analysis process your IDE is running is unaware of the [PIO Construction Environments](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#construction-environments) as documented here and is incorrectly reporting this as a problem.

There may be something you can do to your IDE to make it aware of the PIO Environment - I wouldn't be too concerned.

jake-code, 2020-02-14:
Yeah that's more or less what I was thinking. My concern is the IDE might be doing (or not) something in the background that, although currently, does not seem to interfere with the current code logic might in the future do so with some unexpected results. But anyway, as the build does complete and you do not think it's something critical I guess I can take some time to track down the situation. Do you think uninstalling and reinstalling the Platformio IDE in VSCode might be worth a shot to sort it out?

proddy, 2020-02-14:
@jake-code building the firmware and building the web via the python script are two separate processes. So although you have a fresh .bin file I bet if you removed the /data/www folder it wouldn't be created next time you ask PlatformIO to do a build.

@rjwats with the latest pio 4.2.0 the ldf gets confused with asyncTCP library which is only compiled for ESP32, so I had to add an `lib_ignore = AsyncTCP` to the ESP `esp12e` env. Just a note in case you run into the same issue.

jake-code, 2020-02-14:
I can confirm what prody says about pio 4.2.0 (just upgraded to check) and build does fail with the same code, project and environment that built without problem on previous pio version.

On the subject of the errors on the Problems tab, and picking up on your (rjwats) hint at installed extensions, I removed the installed python extension from Microsoft and the messages went away. Afterwards, I also then noticed that after reinstalling it, the messages appear as soon as I open the script file even without initiating a build. 
So it definitely seems to be related to some pre-check/linting operation that isn't aware of some python functions intrinsic to the IDE. 
As such I guess this issue can be closed if you agree. I would also suggest opening/moving a new one on the pio 4.2.0 problem to log any new developments on the subject.

rjwats, 2020-02-14:
Sure, feel free to raise another issue for the problem with the new PIO
version.

rjwats, 2020-02-14:
PlatformIO should discriminate transitive dependencies defined [libray.json](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/library.json) files by compiling them on the specified platforms only. 

It looks like this is a change in behaviour (by examining the symptom at any rate) here's a [PR](https://github.com/rjwats/esp8266-react/pull/91) which fixes the issue under each currently defined environment.

I think this is a PIO bug.

rjwats, 2020-02-14:
Merged.

rjwats, 2020-02-14:
Feel free to close this @jake-code, unless you need to discuss something else related.

closed 2020-02-14 by jake-code