rjwats/esp8266-react #96: Abillity to monitor/debug RESTful endpoints.

saniyo, 2020-03-16:
Hello, rjwats
With respect to the realized project, I've started to learn it recently and I'm experiencing difficulties incomprehension of the RESTful api. Is it possible somehow to monitor the JSON objects which are available on endpoints directly from the browser by requesting for example /rest/blablabla ? Thanks.

rjwats, 2020-03-16:
You shouldn't have any issues interfacing with the endpoints using SOAP UI,
boomerang or postman for example. Most endpoints are protected with a JWT
Authorization token which you'll have to include with the request as a
request header (see the network log in your browser as you use the app for
some example requests)

To get a token, you can either log in with your browser and get the token
from local storage or log in using your http client software.

saniyo, 2020-03-18:
Thanks, the aforementioned question is solved!

I have some ideas, about interfacing two or more devices in the same network through the interface from one ESP8266. I've had been using for some time this solution https://github.com/tretyakovsa/Sonoff_WiFi_switch. The main backend concept of the project is interacting with similar devices with different interfaces via SSDP just to find each other and provide additional endpoints for the backend which couples devices together under one UI. I wish you had a closer look at it.

closed 2020-05-21 by rjwats