rjwats/esp8266-react #129: Default DEFAULT_BUFFER_SIZE in StatefulService.h may be too small

tichaonax, 2020-05-25:
...esp8266-react/lib/framework/StatefulService.h

#ifndef DEFAULT_BUFFER_SIZE
//#define DEFAULT_BUFFER_SIZE 1024
#define DEFAULT_BUFFER_SIZE 2048 

I am not sure if I increased the buffer size at the right place, I have nested Json object which could not fit into the default buffer size. It took a while the issue was random, it persisted the data but failed to load complaining No Memory. Had to put logging to find the issue.

After the the change I am now good. Let me paste my JSON in case you may want to duplicate.
I am creating a user defined Task Schedule to control for relays. Unless there are other reason for limiting to 1024 I think increasing could be beneficial to others to, then I would not need to redefine the size on updates. The following could not be deserialzed with the 1024 DEFAULT_BUFFER_SIZE

{
    "channelOne": {
        "controlOn": true,
        "name": "Channel One Control",
        "enabled": true,
        "useSchedule": true,
        "schedule": {
            "runEvery": 30,
            "offAfter": 2,
            "startTimeHour": 7,
            "startTimeMinute": 30,
            "endTimeHour": 17,
            "endTimeMinute": 0
        }
    },
    "channelTwo": {
        "controlOn": false,
        "name": "Water Pump",
        "enabled": false,
        "useSchedule": true,
        "schedule": {
            "runEvery": 30,
            "offAfter": 5,
            "startTimeHour": 8,
            "startTimeMinute": 30,
            "endTimeHour": 16,
            "endTimeMinute": 30
        }
    },
    "channelThree": {
        "controlOn": true,
        "name": "Channel Three Control",
        "enabled": false,
        "useSchedule": false,
        "schedule": {
            "runEvery": 30,
            "offAfter": 5,
            "startTimeHour": 8,
            "startTimeMinute": 30,
            "endTimeHour": 16,
            "endTimeMinute": 30
        }
    },
    "channelFour": {
        "controlOn": false,
        "name": "Lawn",
        "enabled": false,
        "useSchedule": true,
        "schedule": {
            "runEvery": 30,
            "offAfter": 5,
            "startTimeHour": 8,
            "startTimeMinute": 30,
            "endTimeHour": 16,
            "endTimeMinute": 30
        }
    }
}

rjwats, 2020-05-25:
https://arduinojson.org/v6/assistant/

You will need around 1345 bytes according to the calculator so yes - you
will need to increase the buffer size.

Having a sensible default which can be overridden was the intention of the
recent change and 1024 bytes is plenty the framework's endpoints so it is a
sensible default.

You can override this without changing the framework code if you stick a
build flag in the platformio.ini file:

-D DEFAULT_BUFFER_SIZE=2048

Alternatively you can override this in the constructor of your own class as
FSPersistence and HttpEndpoint etc expose a bufferSize constructor ardument.

So there are lots of ways to change it for services which require more
memory, I'd personally opt for the latter as it means less memory
allocation for the services which don't require so much memory.

proddy, 2020-05-25:
not to sound like a dick again, but using messagepack when persisting to the FS that example JSON would be only 622 bytes (https://msgpack.org/index.html).

rjwats, 2020-05-25:
Tichaona is talking about the JsonDocument size which is controlled by
DEFAULT_BUFFER_SIZE rather than the size of the string which comes out when
you serialize it. The calculator I posted above shows you the amount of RAM
you would need to hold the data as JsonObejct instances. Using message pack
would not affect the size of the transient JsonObject instances. Unless I'm
totally barking up the wrong tree, in which case please correct me!

Really the setting should probably be called DEFAULT_JSON_DOCUMENT_SIZE -
that might be worth changing.

tichaonax, 2020-05-25:
Thank you I will do like you said.

closed 2020-05-25 by tichaonax