rjwats/esp8266-react #132: Support for back end validtion

tichaonax, 2020-05-26:
I am now able to dynamically create task schedules for fours switches e.g. for water pump, lawn sprinkler etc. Everything is generic you could pretty much use it for anything. User defined Task name and selecting a few choices to set up the task schedule for each switch. When I post a new JSON via postman I want it to persist which is what I get for free by the framework.

There are two things that I want to do and I am not able to figure out if the framework already has some functionality I can use out of the box.

A) When the JSON is persisted whether from form submission or postman the following must happen:

1. Perform some custom validation at the back-end before accepting the JSON for example 
       Running a task very X=30mins amount of time and then turn off after Y=5mins; Y cannot be greater the X. If I use the form submit I could do some validation on the UI. The JSON should be rejected if invalid or violates some rules. Validation at the back end would have it in one place.

2. Reset existing task schedules. I could reset the esp8266 to clear the tasks.

3. Recreate the tasks on startup which is what it currently does. 

B) The UI provides user ability to turn on/off the switch overriding the schedule. If that occurs I want a way intercept that request before processing at the back-end and "suspend" the current task schedule for that switch. When the user switches off or after some predetermined time the current schedule for the switch is restored.

I guess what I am looking for is a way to do some custom validation but still have access to the StatefulService so I can do some needed house keeping.

*NOTE*  In the past I have done these things in .NET C# on a raspberry PI running windows 10 Iot but the OS is very unstable and crashed often and corrupting the SD card. The esp8266 is very promising. Any help or ideas is much appreciated, I will post my code soon when I get to a reasonable state.

rjwats, 2020-05-26:
To solve your event requirement look at adding an observer to the stateful service using addEventHandler. The framework should be able to help you there. Your other issue is interesting and more challanging to solve.

@kasedy had the requirement to want to react to a change in state and only propogate changes if updates are detected during deserialization. This issue is similar, though from a different perspective. Here you want the stateful service to reject an update due to validation constraint(s) being violated (probably resulting in a 400 response if the update is via a REST request).

The major problem with the current design is that it assumes sucessful deserialization will always be possible if default values can be provided where values are not present. It's hard to develop as sophisticated a framework as Spring or .Net under the limited resources of these microcontrollers but this is a missing feature.

This PR shows a possible way of solving this issue:

https://github.com/rjwats/esp8266-react/pull/134

It consists of a minor reafactor in SettingsService which introduces a function which can control processing updates by comparing the current state to the new state if required. The default functionallity has not been altered by this change, so it's only for use when you have more sophisticated requirements (like validation).

So you can override `applyUpdate` in SettingsService and return an outcome to control what happens. You can return:

- OK - To indicate the update has been applied and propagation should take place.
- OK_NO_PROPAGATION - to indicate that the update was fine, but no propagation should happen
- INVALID - to indicate the update was not valid and has not been applied.

I've updated the example to demonstrate how it can be used to only conditionally change the LED state:

```
UpdateOutcome LightStateService::applyUpdate(LightState& newState) {
  if (_state.ledOn != newState.ledOn) {
    _state = newState;
    return UpdateOutcome::OK;
  }
  return UpdateOutcome::OK_NO_PROPAGATION;
}
```

tichaonax, 2020-05-26:
rjwats, this is interesting. I will take a look at the PR you mentioned and merge into my branch so I can take advantage of the new feature. I appreciate your consideration of the issues I raise.

rjwats, 2020-05-26:
No sweat, it's not a given this is going to make it into master so don't
rely on it. It's currently under discussion :)

tichaonax, 2020-05-26:
true, let me know whatever decision is made in the meantime I will explore its potential, I could be an advocate :-)

rjwats, 2020-05-28:
Looking at possibly supporting the following API:

```cpp
  static StateUpdateResult update(JsonObject& root, LightState& lightState) {
    String state = root["state"];
    // parse new led state 
    boolean newState = false;
    if (state.equals(ON_STATE)) {
      newState = true;
    } else if (!state.equals(OFF_STATE)) {
      return StateUpdateResult::ERROR;
    }
    // change the new state, if required
    if (lightState.ledOn != newState) {
      lightState.ledOn = newState;
      return StateUpdateResult::CHANGED;
    }
    return StateUpdateResult::UNCHANGED;
  }
```

This won't facilitate returning meaningful validation results from a RESTful service but it will allow you to (manually) abort an update if validation errors are detected. In the example above, errors in the provided "state" value result in `StateUpdateResult::ERROR` being returned which the framework translates into a 400 response code.

Still a work in progress but it's nearly there: https://github.com/rjwats/esp8266-react/pull/135

This might go some way to satisfying your requirement, of course you could always implmenent your own endpoint code and interface it with StatefulService (see the code in /lib/framework/HttpEndpoint.h for examples). This would give you complete control over the request/reponse handling and you could make it do whatever you want.

tichaonax, 2020-05-28:
cool

tichaonax, 2020-05-28:
before I go too far in trying to create my own endpoint I wanted to take as much advantage as possible from the framework

rjwats, 2020-05-29:
The latest changes in master allow you to perform validation and reject updates if they don't meet the right criteria. There's are some example in the README, hope that can work for you.

tichaonax, 2020-05-30:
Thank you for finding this solution

closed 2020-05-30 by rjwats

tichaonax, 2020-06-01:
> The latest changes in master allow you to perform validation and reject updates if they don't meet the right criteria. There's are some example in the README, hope that can work for you.

Hi, I did verify that back-end validation works both via form and postman. However while the data is not updated if invalid, when using forms the notification still says Saved Successfully. I would like that to be a Warning or Error message saying updates where not applied. Better is there a way I can override that message and issue another with a more appropriate custom message.

rjwats, 2020-06-01:
RestController should respond to any non 200 response code by displaying a
snackbar notification with a warning. Can you post your back-end code? Are
you returning ERROR or UNCHANGED?

UNCHANGED will respond with a 200 response code, ERROR with a 400.

tichaonax, 2020-06-01:
I see l was returning UNCHANGED, l think if l return error should fix it. I will test and see, thanks again.

tichaonax, 2020-06-01:
I changed to ERROR and now gets a 400 which is OK for rest API but for UI it stops the UI from working instead it shows a continous spinner. I need to figure out how my controller should respond to a 400. Basically I want the error message but reload original settings user acknowledgment

tichaonax, 2020-06-01:
It might be  bit difficult to solve if the UNCHANGED actually saves the old data as unchanged but the behavior is correct on the UI it does not accept the new changes but the misleading snack message if the user does not actually look at the screen.

rjwats, 2020-06-01:
Currently, it is your responsibility to write the update routine in a way
where it doesn't update the value unless the operation is going to be
successful.

The code in APSettingsService demonstrates this approach:

  static StateUpdateResult update(JsonObject& root, APSettings& settings) {
    APSettings newSettings = {};
    newSettings.provisionMode = root["provision_mode"
] | FACTORY_AP_PROVISION_MODE;
    switch (settings.provisionMode) {
      case AP_MODE_ALWAYS:
      case AP_MODE_DISCONNECTED:
      case AP_MODE_NEVER:
        break;
      default:
        newSettings.provisionMode = AP_MODE_ALWAYS;
    }
    newSettings.ssid = root["ssid"] | FACTORY_AP_SSID;
    newSettings.password = root["password"] | FACTORY_AP_PASSWORD;
    if (newSettings.provisionMode == settings.provisionMode && newSettings.
ssid.equals(settings.ssid) &&
        newSettings.password.equals(settings.password)) {
      return StateUpdateResult::UNCHANGED;
    }
    settings = newSettings;
    return StateUpdateResult::CHANGED;
  }

APSettings is a trivial class which supports copy assignment so its easiest
to craete a working copy of APSettings, update it and then assign it at the
end with:

```cpp
     settings = newSettings;
```

rjwats, 2020-06-01:
Currently, it is your responsibility to write the update routine in a way
where it doesn't update the value unless the operation is going to be
successful.

The code in APSettingsService demonstrates this approach:
```cpp
  static StateUpdateResult update(JsonObject& root, APSettings& settings) {
    APSettings newSettings = {};
    newSettings.provisionMode = root["provision_mode"
] | FACTORY_AP_PROVISION_MODE;
    switch (settings.provisionMode) {
      case AP_MODE_ALWAYS:
      case AP_MODE_DISCONNECTED:
      case AP_MODE_NEVER:
        break;
      default:
        newSettings.provisionMode = AP_MODE_ALWAYS;
    }
    newSettings.ssid = root["ssid"] | FACTORY_AP_SSID;
    newSettings.password = root["password"] | FACTORY_AP_PASSWORD;
    if (newSettings.provisionMode == settings.provisionMode && newSettings.
ssid.equals(settings.ssid) &&
        newSettings.password.equals(settings.password)) {
      return StateUpdateResult::UNCHANGED;
    }
    settings = newSettings;
    return StateUpdateResult::CHANGED;
  }
```
APSettings is a trivial class which supports copy assignment so its easiest
to craete a working copy of APSettings, update it and then assign it at the
end with:

```cpp
     settings = newSettings;
```

tichaonax, 2020-06-02:
Yes I did implement it that way but unless you return StateUpdateResult::ERROR; the rest controller will give misleading message. I would like to suggest that the response to the UI from controller include a new property with two fields the StateUpdateResult and a custom text field user can return as part of the back-end validator. With this information the UI can react to that. Also the rest-controller should not dispatch the save successifull snack message in that case.

The 400 is good way also except that UI ends up just showing a spinner and nothing loading as the data state is at that point undefined. Just suggestions, either or both of addresses to these two will put us at a very good sport for the framework