rjwats/esp8266-react #139: RestController Support: Updating a single value in nested JSON

tichaonax, 2020-05-30:
[image]

[image]

The RestController works great for root level attributes. I ran into an issue when wanting to update the runEvery nested attribute. I had to cheat as you see in the image writing back to the event object by calling my own form event handler and repackage the event object to play nice with the RestController handleValueChange method as it can accepted a single object.

Its a nice to have but for now I do have a workaround.

rjwats, 2020-05-30:
I probably would have chosen to do that slightly differently. You could write a simple wrapper which uses the extractEventValue helper which would work for any of your schedule values:

```typescript
      handleScheduleValueChange = (name: keyof Schedule) => (event: React.ChangeEvent<HTMLInputElement>) => {
        const { data, setData } = props;
        setData({ ...data, schedule: {...data.schedule, [name]: extractEventValue(event) } });
      }
```

tichaonax, 2020-05-30:
But I do not want to modify the framework files. Your solution seems to suggest I put my change there which could be overwritten with your future updates. My workaround is all in my code that I control. Unless I am missing something. I like your suggestion however.

tichaonax, 2020-05-30:
OK I see what you mean now that is my callbak func, I will use that approach instead.

tichaonax, 2020-05-30:
That worked beutifull, I am using a Select box so had to change
(event: React.ChangeEvent<HTMLInputElement>) => 
  to
(event: any) =>
This is exactly what I wanted, thanks for a quick suggestion and better solutoon.

closed 2020-05-30 by rjwats