rjwats/esp8266-react #141: Store logs

uncodead, 2020-05-31:
I'm facing problems when store a large amount of log (temperatures) in SPIFFS.

Is there another way to store this data?

saniyo, 2020-05-31:
I suppose to connect a microSD card for that purpose.

rjwats, 2020-05-31:
Hey Bruno,

Let's have a bit more information. What type of data are you storing? Is it
variable length (does it contain strings)? How often are you logging and
how much are you storing. Are you storing serialized JSON?

This is a tricky problem really, one which I solved in the past with a
circular file system buffer. Fixed length data means it's easier to seek to
the right location in the file which makes things much easier.

 I was never pleased with that solution. I'm working on storing logfiles
(for debugging) on the filesystem and wonder if a generalized solution
might be in order.

rjwats, 2020-06-11:
Let me know if you still need help with this Bruno - reopen if I can help :)

closed 2020-06-11 by rjwats