rjwats/esp8266-react #144: How update state from Timer job

tichaonax, 2020-06-04:
This might simple. I have a StateFull class that initializes a Timer. That Timer periodically wakes up turns on/off I/O pin. How do l pass back that state change so that other parts of the system gets the state change. At the moment the only way appears to be making an http request to an endpoint to update the state but that appears expensive and requires authorization first. Does anyone have a better alternative.

rjwats, 2020-06-04:
Unless I've misunderstood what you are trying to achieve (it would be
easier to understand if you shared some code), StatefulService should
facilitate what you are trying to do.

You can store the state of the pin in the state object and use the normal
observer/propagation mechanism it provides to alert the rest of the system.
I think there's an example just like this in the README?

tichaonax, 2020-06-04:
#140 .....
class ChannelStateService : public StatefulService<ChannelState> {
 public:
  ChannelStateService(AsyncWebServer* server,
                    SecurityManager* securityManager,
                    AsyncMqttClient* mqttClient,
                    //LightMqttSettingsService* lightMqttSettingsService,
                    FS* fs,
                    TaskScheduler taskScheduler,  //  scheduler instance
                    int channelControlPin,  // 5
                    char* channelJsonConfigPath,  //  "/config/channelOneState.json"   
                    String defaultChannelName,  //  Channel One Control
                    String restChannelEndPoint, //  "/rest/channelOneState"
                    char* webSocketChannelEndPoint //  "/ws/channelOneState"
                    );
..........
The TaskScheduler taskScheduler,  //  scheduler instance
is then used create the job for this single pin

void ChannelStateService::begin() {
    _state.channel.controlPin = _channelControlPin;
    _state.channel.controlOn = DEFAULT_CONTROL_STATE;
    _state.channel.name = _defaultChannelName;
    _state.channel.channelEndPoint = _restChannelEndPoint;
    _fsPersistence.readFromFS();

    _taskScheduler.setScheduleTimes(_state.channel);

    onConfigUpdated();
}
void ChannelStateService::onConfigUpdated() {
    _state.channel.controlPin = _channelControlPin;
    digitalWrite(_state.channel.controlPin, _state.channel.controlOn ? CONTROL_ON : CONTROL_OFF);
}

In the scheduler once the jobs are set the following function are periodically called:

void TaskScheduler::scheduleTask(){
  Alarm.timerRepeat(_channel.schedule.runEvery, std::bind(&TaskScheduler::runTask, this));  // periodic schedule
  Serial.print("Schedule Task for channel : ");
  Serial.println(_channel.name);
  Serial.println("Control Scheduler started => ");
  runTask();  // run once initially on set
  digitalClockDisplay();
}
void TaskScheduler::controlOn(){
  Serial.print(_channel.name);
  Serial.print(": Task started at: ");
  digitalClockDisplay();
  digitalWrite(_channel.controlPin, CONTROL_ON);
  Alarm.timerOnce(_channel.schedule.offAfter, std::bind(&TaskScheduler::controlOff, this)); 
}

void TaskScheduler::controlOff(){
  digitalWrite(_channel.controlPin, CONTROL_OFF);  
  Serial.print(_channel.name);
  Serial.print(": Task stopped at: ");
  digitalClockDisplay();
}

What I want is that when these last two functions toggle the pin the state of the PIN I what that reported back back to the StateFull service. Note TaskScheduler is instance passed to the Service. In other words an outside process is changing the state of the pin.

Everything works very well except I do not know how to sent back unless I make a httprequest to the endpoint with the state change and even that one how to do it without authentication I get 401 right now coz I m not passing the bearer token

tichaonax, 2020-06-06:
Basically if it can an be done the requirement is: Provide a way of registering a state change from an external event.

rjwats, 2020-06-06:
Updating the state managed by a StatefulService instance should be done
with the callback mechanism in order to propagate the changes using the
callback notification approach.

There is an example in the README which shows this, here is an excerpt. Any
class which wishes to observe an state change can register a update handler:

```
update_handler_id_t myUpdateHandler = lightStateService.addUpdateHandler(
  [&](const String& originId) {
    Serial.print("The light's state has been updated by: ");
    Serial.println(originId);
  }
);
```

Any any update to the state should be done using the update callback:

```
lightStateService.update([&](LightState& state) {
   if (state.on) {
    return StateUpdateResult::UNCHANGED; // lights were already on, return
UNCHANGED
  }
  state.on = true;  // turn on the lights
  return StateUpdateResult::CHANGED; // notify StatefulService by returning
CHANGED
}, "timer");
```

Your code looks like it's manipulating the state directly which isn't
really the intention when using StatefulService. StatefulService does
expose a function called "callUpdateHandlers" if manual invocation of the
update handers is required, this would be for special cases though and the
callback approach is advisable.

tichaonax, 2020-06-06:
Thanks for the feedback. I did implement it per the example and that part is working correctly. If you change the state via controller or postman it's picked up by the other in this case the web socket. If the websocket controller makes a change it also propagates this is exactly as you have shown in examples. What is missing is that in side my statefulcontroller l start some other "process" using a Timer that one can turn on the switch directly, l want to do that through the state and l don't know how to do it. Yes l am manipulating the state directly in that case. It is probably a special case. May be you can help show me how l make a restful API call with a beartoken if not how can l make a non-expiring bearertoken if what l am saying makes sense at all.

rjwats, 2020-06-06:
When you say "some other process" do you mean an external process? Such as
a service running on a server? Or a timer running on the device?

If it is a timer running on the device, it should be as simple as the
example in my previous email. Just update the state by calling update()
with a callback. If it's an external system, such as a process running on a
server, then yes, you'll have to update the state through HTTP/WebSockets
or MQTT as you described.

tichaonax, 2020-06-06:
It is a Timer, let me look at your previous email I think you are right, thanks.

tichaonax, 2020-06-07:
I think I am completely lost trying to go with the method described above, how do you invoke those callback methods from say my Timer functions. do you have a concrete example. I am trying to call either the read or the update method but I get the following:

no instance of overloaded function "LightStateService::read" matches the argument list -- argument types are: (LightState)

rjwats, 2020-06-07:
You might make better sense of this if you look at some of the XXXSettingsService classes in the framework for an example. Most of these classes register observers (update handlers) in their constructors. For example in WiFiSettingsService.cpp:

```cpp
addUpdateHandler([&](const String& originId) { reconfigureWiFiConnection(); }, false);
```

Now whenever an update is made to the state that WiFiSettingsService oversees, reconfigureWiFiConnection is called and can respond to the change. Anywhere you have a reference to the WiFiSettingsService instance you can call its update function (with a callback - not a state object) to safely update its state and propagate the change:

```cpp
wifiSettingsService.update([&](WiFiSettings& wifiSettings) {
  wifiSettings.ssid = "MyNetworkSSID";
  wifiSettings.password = "MySuperSecretPassword";
}, "some_origin");
```

This could be done in a timer function too for example. What you are trying to do should be simple, but I can't see your project - so it's hard to be sure.

tichaonax, 2020-06-07:
Ok thanks, l will push my code base soon so you can see it. I think l am beginning to understand. Thanks again when l push my repo you can assist further.

tichaonax, 2020-06-07:
OK I think I implemented as you said but I get the following error message on the two Timer functions I am calling update from:
 'void TaskScheduler::controlOn()': no matching function for call to 'ChannelStateService::update(TaskScheduler::controlOn()::__lambda6, String&)'
           }, _channel.name);

I have pushed the code to: 

https://github.com/tichaonax/esp8266-react-scheduler

When you get a chance please can you tell me why its not working, thanks!

rjwats, 2020-06-07:
Your update function does not match the signature because it does not
return an instance of StateUpdateResult. There's an example in the readme.

tichaonax, 2020-06-08:
Are you saying ControlIOn() must return StateUpdateResult or the update function that reads/writes to the JSON object.

tichaonax, 2020-06-08:
Man it really works like magic, this saved me a whole lot of bad design I was about to do. This is really good stuff you did, now let me concentrate on my functionality now that the framework support is all good.

closed 2020-06-08 by rjwats