rjwats/esp8266-react #146: Front end compilation throw out an error

saniyo, 2020-06-08:
I've decided to check the new features I made a cloned copy of the pool. During TS compilation got an error:
 >C:/Users/saniy/Documents/PlatformIO/Projects/react-esp8266->latest/latest/interface/src/components/RestController.tsx
>TypeScript error in C:/Users/saniy/Documents/PlatformIO/Projects/react-esp8266->latest/latest/interface/src/components/RestController.tsx(103,11):
>'handleValueChange' is specified more than once, so this usage will be overwritten.  TS2783

 >  101 |       render() {
 >   102 |         return <RestController
 > > 103 |           handleValueChange={this.handleValueChange}
 >       |              ^
 >   104 |           setData={this.setData}
 >   105 |           saveData={this.saveData}
 >   106 |           loadData={this.loadData}

> error Command failed with exit code 1.
> info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

what should I do?

rjwats, 2020-06-08:
This is a typing issue in the controller which is a bit tricky to solve
properly - but you can make the TS compiler be quiet by re-ordering the
parameters as shown in this commit:

https://github.com/rjwats/esp8266-react/pull/143/commits/1bcf61e0973e0a63bd339926660be28912ff6204

Will work on a proper fix to these components as part of this next PR.

rjwats, 2020-06-08:
Not it affects both the WS and HTTP components

saniyo, 2020-06-08:
Quite strange, I thought it is "compileable" out from the box on the master's origin. Does this issue is going to be solved in further commits? In your opinion will the Linux build eliminate it? I really don't want to bypass any possible problems.

saniyo, 2020-06-08:
Having moved up the WebSocketController and RestController props allowed me to solve the issue. The second time, I would make the TS compilation be quite, but some times it makes new troubles. Thanks.

rjwats, 2020-06-08:
It looks like a change in the TypeScript compiler during a minor version
bump has caused this (though it's really a problem with the interface code).

Will push an update into master to fix the issue, then resolve it properly.

closed 2020-06-08 by rjwats