rjwats/esp8266-react #149: Kudos!

Rick-OGSS, 2020-06-11:
Hey RJ,

Just a quick note to say thanks for sharing this repo.  I was about 20% done almost exactly the same project when I came across yours and got totally stoked on your work!  You have saved me hundreds of hours of keyboarding and I have learned some some cool things about working with state and MQTT too.

How can I buy you a keg of beer?
Paypal?

Cheers,

Rick

rjwats, 2020-06-11:
I'm glad you could make use of the framework Rick. I wanted something which would allow me to have an extremely configurable user interface for my projects and this came out of it. It's not perfect and it requires "full stack" knowledge but I think it's a decent starting point for IoT projects if you're willing to put in some work.

I'm very grateful of the appreciation. I do have a [paypal link](https://www.paypal.com/cgi-bin/webscr?cmd=_donations&business=F5UZQ4NWFNGMJ&currency_code=GBP&source=url) if you feel compelled, the kind thanks are more than enough though :)

closed 2020-06-11 by rjwats

rjwats, 2020-06-12:
Thanks so much man, it's really generous of you.

I'll order some delicious hops and grain for my next brew :)

Rick Watson