rjwats/esp8266-react #151: Easly Identify My Devices Nice To Have

tichaonax, 2020-06-13:
As a user controlling several devices on my phone I want to be able to know which device I am dealing with without too much navigation. It would be nice to have either the host-name/IPaddress somewhere in the navigation bar and or the signout box. Add option to disable that for those not wanting to see that information. The WIFI settings form could have a check box to enable the feature as this is where the host-name can be edited. This is the domain of the Framework code so I cannot make changes to except ask if it could be accommodated. Please find mock screens of my suggestions off my current UIs.
[image]
[image]

saniyo, 2020-06-13:
I've already mentioned something similar to your needs, I've been using this https://github.com/tretyakovsa/Sonoff_WiFi_switch solution for some time. The key feature of it is in using the SSDP protocol for similar devices in the same network. The back-end of them finds each other and sends data about the available device and its IP address, so then the front-end from each of them can unite sensors from different devices under one UI. It would be good if the developers could have a look at it. It would help to scale this project horizontally.

tichaonax, 2020-06-13:
Thanks, l will take a look at it.

rjwats, 2020-06-13:
I agree that it would be nice to have a little more status information "up front" in the UI. In fact there's already a ticket open to that effect: #104 

The way I saw it, icons indicating the status of the access point and WiFi connection would be the most useful things to have in the header. Icons communicating:

- Is the access point on the device active, and what # clients does it have?
- Does the device have a WiFi connection, and to which SSID & with which hostname?

The header could summarize the status with a color/icon/badge and the extra information being displayed on click with a popover.

On the topic of having a unified UI for your devices, personally I use MQTT and HomeAssistant for such things because it's centralized and will grow as your needs do. It's really neat and very well supported by the community.

tichaonax, 2020-06-13:
You are correct, it's definitely a nice to have. I will track #104 and see when it makes it to master. I will be incorporating home assitant at a later stage, l wanted this to be able to work kind of independently especially in remote places where internet is sporadic or non existent. To that a ui to set local time in the event of no internet access would be also very good as the device can be completely be operated with its own access point.

rjwats, 2020-06-13:
Yeah, setting local time manually should be on the list, can you raise a
separate ticket for that?