rjwats/esp8266-react #152: Setting Local Time Manually

tichaonax, 2020-06-13:
This is something really important especially for my application. Where internet is intermittent or non existent the device Access Point is good enough to operate the device. Ability to set local time becomes critically important. At the moment I hot-spot to my cell phone then once the schedules are set the device is good to go as long as the device is not reset.

rjwats, 2020-06-15:
Thx. Yes, this is a missing feature - will get this implement this week.

rjwats, 2020-06-15:
PR needs a review and a tidyup tomorrow. Looks decent though:

[image]

[image]

rjwats, 2020-06-15:
NB - using browser "datetime-local" editor support rather than the MUI pickers to save space on the device. This is also up for review but i think the [MUI date pickers ](https://material-ui-pickers.dev/)are probably a little too large (18k minzipped) to include for this feature.

tichaonax, 2021-02-18:
I wanted to say we got hit by bad cold front Sunday night and went without electricity or wifi for more than 72 hours. Since NTP was inactive, I was not able to setup as the setup button did not get enabled. I tried to disable NTP service but it did not show. My esp8266 were connected to a router that did not have an uplink. Local network worked. What are the exact conditions under which the set-time button is enabled.

rjwats, 2021-02-18:
The set time button should be enabled whenever the NTP is disabled (the
checkbox on the settings form)

rjwats, 2021-02-18:
It uses the underlying enty_enabled flag to determine whether or not to allow the user to set the time manually - which seems to make sense is suppose because otherwise the manually set time would be overridden by the next NTP sync event. 

Whenever the device is disconnected from WiFi, or if NTP is explicitly disabled it should become automatically available. But of course if ntp is not being correctly disabled by the SettingsService code this could fail... Here are some of the important bits:

[image]
[image]
[image]

(And basically the whole of NTPSettingsService)

Can you re-produce the problem with a series of steps I can re-create here?

tichaonax, 2021-02-26:
Sorry in delaying my response. You are right it did show the set time button but l needed to disable ntp first. I guess my situation was a bit different l was expecting the button to be enabled for a situation like what happened to me.. Having the ntp override my time is actually preferred in my case but l may just want to set the time if ntp is not available for whatever reason but might be available later. My systems will not enter their routines until a valid date time. I think having the button enabled all the time is fine it will still work for those who have ntp disabled and if l do not want override then l can disable ntp altogether but to only allow set time when ntp is disabled would be a loss of functionality l think. Otherwise it is wolike you said.

tichaonax, 2021-02-26:
The steps to reproduce would be. Have the esp device connected to wifi router. Disable the router uplink and reset the esp device.  Ensure esp device still connected to wifi, If the ntp is enabled the set time won't show. I expected the button to show l think in my case l want it enabled all the time.

rjwats, 2022-03-18:
Yeah, I follow what you're suggesting and it's a bit of an edge-case but I suppose still valid.

I guess there's no harm in enabling the set time feature at all times (could remove the check in the UI yourself). If making this change for everyone I'd put a warning in the dialog if NTP is enabled to say somthing like: "NTP is enabled, the set time you are setting will be overwritten when NTP next synchronises"