rjwats/esp8266-react #155: JWT with RSA algorithm

uncodead, 2020-06-17:
Is it possible to make JWT with RSA algorithm (public and private keys)

rjwats, 2020-06-17:
> Is it possible make JWT with RSA algorithm (public and private keys)?

I'm sure it's possible - though in my understanding RSA is way more memory
hungry - so that will be interesting.

Out of curiosity, what is the requirement?

uncodead, 2020-06-17:
Avoid secret keys inside project... we'll have only a plublic key inside project.

Thks

rjwats, 2020-06-17:
The secret can be overridden on the device once flashed and by default it
now uses the chip id (different devices get different-ish default secrets).
So it's not awful in that regard.

I think without SSL on the HTTP endpoints security is somewhat flawed
anyway. When I last looked at this it looked unfeasible as TLS required
~15k of RAM per request at a minimum.

I think the security can be described as "best effort", not "enterprise
grade". Designed to stop the casual person who discovers the devices on the
network :)

On this subject, I've been pondering if JWT expiry and a token generator in
the UI might be worth implementing though. So you can generate tokens for
interfacing with the device externally.

uncodead, 2020-06-18:
JWT expiry is the best for now...

uncodead, 2020-06-19:
Changing the subject, have you managed to make https requests on esp8266? I am suffering!

rjwats, 2020-06-19:
It's not something I've tried. I did try to get TLS working on MQTT... it
"worked" but was really impractical due to the memory requirements.

I'm sure you'll have similar problems with https, it's probably possible if
it's just about the only thing the device is doing.

What are you trying to do exactly?

uncodead, 2020-06-19:
i need to call an api (https://brewuno.com/api/) but inside my project it isnt connecting.
when i create a project from scratch works. i'm supposing is a issue inside wifisettins.cpp.

rjwats, 2020-06-20:
Pretty sure you need to be thinking about this as a memory issue:

- Getting HTTP working seems to be trivial in a loop() function
- When switching the same code to use HTTPS - using BearSSL::WiFiClientSecure as the client the client returns "connection refused"
- I managed to get HTTPS "working" after stripping features out of the framework till there was enough memory available to service the request

It looks like you'll need about @ 30k of free heap to make this work:

[image]

During the request the available heap drops significantly:

[image]

Adding some memory metrics to [this example](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/examples/BasicHttpsClient/BasicHttpsClient.ino), you can see that the HTTPS request is incredibly memory hungry:

[image]

So yeah... it needs 28,816 bytes for the request!

rjwats, 2020-06-20:
I can also confirm I have no issues connecting to HTTPS services on my ESP32 boards - more RAM available here:

[image]

uncodead, 2020-06-21:
it was a memory issue! thanks!

closed 2020-06-21 by uncodead