rjwats/esp8266-react #156: development simulator options

Rick-OGSS, 2020-06-19:
I am using the ESP32 and building out your framework as a complete controller with UI.  Right now I am focusing on the React interface but am finding the dev process slow as I edit=>save=>build=>upload to ESP=>log in then test. (your python script really helps here)

When I run node locally (npm start) I can only get to the login screen which makes sense as the arduino code is not serving up its portion.

Is there some method you know of that would allow us to somehow simulate the ESP portion of the project for local testing, without having to download and connect to test?

Thanks!

Rick

rjwats, 2020-06-19:
Hi Rick,

There's a section in the docs which talks about the way you might approach
this:

https://github.com/rjwats/esp8266-react#running-the-interface-locally

If you have to upload the device for every interface change the turnaround
time makes it very time consuming, and the guide above shows how you can
deploy the backend to the device and work on the interface locally. That
way you only need to upload the device during development if you make
backend changes.

Based on the guidelines posted above, my normal development process looks a
bit like this:

   - Develop back-end changes
   - Enable CORS in platformio.ini
   - Deploy back-end changes to device
   - Configure endpoint root to point to device IP address in
   interface/.env.development
   - Run interface locally with "npm start"
   - Develop interface locally

This seems to work OK for me most of the time, however at one point during
the early development I had a node app which would serve up simulated
responses for the various endpoints but this was challenging to keep in
sync with the changes as the app developed. Now that things are more stable
it might be worth re-introducing if it could be useful. Personally I've not
really missed it when working as described above.

rjwats, 2020-06-19:
Hi Rick,

There's a section in the docs which talks about the way you might approach this:

https://github.com/rjwats/esp8266-react#running-the-interface-locally

If you have to upload the device for every interface change the turnaround time makes it very time consuming. The guide above shows how you can deploy the backend to the device and work on the interface locally. That way you only need to upload the device during development if you make
backend changes.

Based on the guidelines posted above, my normal development process looks a bit like this:

- Develop back-end changes
- Enable CORS in platformio.ini
- Deploy back-end changes to device
- Configure endpoint root to point to device IP address in interface/.env.development
- Run interface locally with "npm start"
- Develop interface locally

This seems to work OK for me most of the time, however at one point during the early development I had a node app which would serve up simulated responses for the various endpoints but this was challenging to keep in sync with the changes as the app developed. Now that things are more stable it might be worth re-introducing if it could be useful. Personally I've not really missed it when working as described above.

rjwats, 2020-06-19:
Have just re-read the docs. I think it they should be more explicit about the steps involved here.

The critical details about the configuration of the endpoint root and CORS are hidden away at the end and easy to miss. I'll update this later, so it makes more sense.

Rick-OGSS, 2020-06-19:
Great, I will give it a try and let you know.

Rick-OGSS, 2020-06-19:
That did the trick!
Your bullet point list really cleared it up and made the process easy.  I recall reading that section but I missed the gist of it originally.
Cheers

closed 2020-06-19 by rjwats