rjwats/esp8266-react #160: Feature Request: OTA updates over Web/HTTP in UI

proddy, 2020-06-22:
I useful feature I believe would be able to upload a firmware .bin from the WebUI, so OTA over HTTP. It's fairly trivial to implement and plenty of examples around (e.g. https://github.com/ayushsharma82/AsyncElegantOTA/blob/master/src/AsyncElegantOTA.h). I may have a go myself when I port my projects over to the latest esp8266-react

rjwats, 2020-06-22:
I agree this would be handy. I also like the idea of triggering the device
to pull an update over HTTP (supported by github's release API in the UI).

As part of this, I think I need to do the work required to get the
framework using a proper versioned release strategy so it can support the
feature properly.

proddy, 2020-06-23:
For versioning you could look at [Semantic Versioning](https://semver.org/spec/v2.0.0.html) and keeping a [ChangeLog](https://keepachangelog.com/en/1.0.0/).

Pulling an update from GitHub would a lovely addition, but not for everyone as it requires setting up a CI/build pipeline like Travis, CircleCI). I've coded this before so will give it a go on esp8266-react when I've finish the port. 

(BTW since the framework also supports ESP32 (and soon ESP32-S2) maybe a good time to rename the project before you can swamped with a zillion followers?)

rjwats, 2020-06-27:
Hey Paul,

This looks to be working quite well now in the feature branch (see PR). I've tested it on various ESP32 and ESP8266 devices and all seems good. I'm still reviewing the changes but feel free to take a look and see how it works for you:

[image]

[image]

Implementation notes:

No Axios in the dependency un UI ATM so I had to use XMLHttpRequest (yuck) to get upload progress working. 

The backend is more defensive than most implementations I've seen. I don't think most implementations take into account the possiblity of simultaneous requests. It's an unlikely scenario, but the async webserver allows for it so...

Things I'm still thinking about:

- Effect on react-dropzone on bundle size when minzipped. I think it's very small but need to check.
- Is this feature considered a separate feature to normal "OTA" updates or is it controlled by the same FT_OTA feature flag? I think it probably needs it's own flag as not everyone will want it.
- Does "Update" require it's own top level menu item or is it OK under "System"? I think it's probably OK where it is for now, maybe when I look at HTTP pull from git.

proddy, 2020-06-27:
Hi Rick, that was quick! I had a quick try on an ESP8266 and it works wonderfully well. Some initial observations and feedback to your questions:
- I wouldn't call it OTA Upload. OTA is bit a techie abbreviation. Perhaps just "Firmware Update" ?
- yes I would definitely secure it under its own feature flag, to stop people bricking the esp with some trojan code
- It does belong under System, so where it is now is perfect I think
- Would be great if you could add a Cancel button in case you want to panic-quit the upload
- It would be nicer to show a % progress instead of the number of bytes. If you really want to show bytes, round it to KBs. Displaying a number like 1,444,760 is a bit daunting.
- The open file dialog on Windows shows "Custom Files (.exe,.com,.bin)". You may want to restrict it only to `.bin` files.
- I did try and upload some junk .bin files and it does show an error, I presume from the unpacking. The error is `Problem uploading: Invalid status code: 500`. Is there a way to catch this and show a more meaningful message like `Upload failed.`?
- Later perhaps modify the text "_Upload the new "firmware.bin" file using the form below. This will immediately start the firmware replacement._" as sometimes the firmware may not be called `firmware.bin` and the word 'form' is confusing to non-techies.

I started today with my port to esp8266-react so keeping a note of findings along the way which I'll share later. The project has come along way since last December, great progress & thanks

rjwats, 2020-06-27:
Thx, some good points there:

- **Perhaps just "Firmware Update"** - Yes, I agree OTA Upload is a confusing name for the feature. We want to make sure there's cohesion with the possible "Download Firmware" feature... "Upload Firmware" maybe?
- **Cancel button** - Yes, I had assumed this would be useful and considered it too - will implement.
- **It would be nicer to show a % progress instead of the number of byte"** - agreed, the exact number of bytes isn't particularly useful.
- **You may want to restrict it only to .bin files.** - Yes, I found it tricky to get react-dropzone to behave perfectly, there are known issues with it's mime type detection and using ".bin" as the accept prop didn't seem to apply correctly to the drop behaviour (only the dialog). I'll have another look and see if it's possible to do anything more elegant.

- **show a more meaningful message** -  Yes I thought about this - It's possible to get error codes out of the update instance. While it's easy to return the code in the body of the response and translate it in the UI (I didn't want to send the string in case someone suggests i18n is an important feature), it might be a bit more effort than it's worth.

rjwats, 2020-06-27:
"Upload a new firmware (.bin) file below to replace the existing firmware."

Better?

rjwats, 2020-06-27:
**Pushed changes:**

- Renamed feature to "Upload Firmware"
- Added cancel button
- Showing % instead of absolute progress values
- Using meaningful message on upload screen
- Separated setting: "FT_UPLOAD_FIRMWARE"

**Not done:**

- May come back and improve the generic 500 response later
- Had another look at dropzone mime type / file dialog behaviour - annoyingly tricky.

proddy, 2020-06-27:
nice. Just tested and looks all good.

closed 2020-06-28 by rjwats