rjwats/esp8266-react #161: Consant crashing accessing webserver, can't figure out

tichaonax, 2020-06-24:
[image]

I have the latest master merged in my branch. Works well until I try to access the webpages. I have looked for potential watchdog timer issues but cannot find anything. I have included two exception dump files to see if that can help. They alternate between Exception (28) and Exception (29) consistently.

Exception (29):
epc1=0x4020da96 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
  epc1=0x4020da96 in AsyncWebServerResponse::AsyncWebServerResponse() at ??:?

The code base is on github:

https://github.com/tichaonax/esp8266-react-scheduler/
[crash-dump-01.txt](https://github.com/rjwats/esp8266-react/files/4825347/crash-dump-01.txt)
[crash-dump-02.txt](https://github.com/rjwats/esp8266-react/files/4825349/crash-dump-02.txt)

rjwats, 2020-06-27:
Soz I've been super busy and haven't managed to pull your code and run it up yet. Not sure if I'll find the time till later this week TBH.

I've seen these types these types of errors before (Exception3, Exception 9, Exception 28 and Exception 29) when memory is low. 

Also I encountered these types of problems when logging to Serial during callbacks from the AsyncWebServer. Not much to go on for now, but it might give you some clues.

tichaonax, 2020-06-27:
rjwats, you are correct its memory. Do not waste anymore time on this it can be closed. I reduced my timers and UI one by one and both the web server work without hitches only for two schedules. Fuur schedules work but without access to the web-server. It cannot handle both. I really wanted the four channels on one chip but looks like I will have to have two chips.

rjwats, 2020-06-27:
You could consider esp32 too, lots more memory.

tichaonax, 2020-06-28:
It was running out of memory, had to reduce the number of timers to one for it work without any hitch

closed 2020-06-28 by tichaonax