rjwats/esp8266-react #163: Need support for NTPStatus notifiction

tichaonax, 2020-06-28:
Need a way to hook up to a NTP status change either via NTP or user changing the local time with the new UI. Time change requires reconfiguring scheduling algorithms as I would like to reset them each time the local time is updated.

Something similar to the following would be great.

esp8266React.getWiFiSettingsService()->addUpdateHandler(
  [&](const String& originId) {
    Serial.println("The WiFi Settings were updated!");
  }
);

rjwats, 2020-06-28:
This may be tricky, I briefly looked into it when I did the sntp stuff:

- ESP32 - I know IDF supports **sntp_set_time_sync_notification_cb** but this isn't exposed by the Arduino wrapper yet
- ESP8266 - I don't know of any API which lets you listen to the sntp sync event

tichaonax, 2020-06-28:
Ok, I will implement some polling mechanism, thanks for looking into it.

closed 2020-07-25 by rjwats