rjwats/esp8266-react #165: Make access point more configurable

rjwats, 2020-06-28:
The access point currently gets the default IP address configuration:

- **IP** 192.168.4.1
- **Gateway** 192.168.4.1
- **Subnet** 255.255.255.0

It may be desirable to make this configurable in APSettingsService. Other configuration options could include activating/deactivating the captive portal provided by DNSServer, though this is less important.
 
```cpp
  IPAddress localIP(192, 168, 4, 1);
  IPAddress gateway(192, 168, 4, 1);
  IPAddress subnet(255, 255, 255, 0);
  WiFi.softAPConfig(localIP, gateway, subnet);
```

proddy, 2020-07-02:
I would say nice to have. People are used to 192.168.4.1 as the default AP IP, but then again it's pretty simple to change.

rjwats, 2020-07-02:
I have this ready in a local branch :)

closed 2020-07-07 by rjwats