rjwats/esp8266-react #169: Replace moment with date-fns

rjwats, 2020-07-07:
Could save about 20k if we switch to date-fns as it's much more tree-shakable. It's only the parser/formatter we use from moment IIRC.

tichaonax, 2020-07-11:
This is great, I was already using it.

rjwats, 2021-01-17:
Luxon is a more suitable replacement for the framework. It supports tree shaking so results in a smaller artefact. And a key feature which is missing from date-fns but is present in luxon is the ability to reatin the offset when parsing ISO 8601 strings:

https://github.com/date-fns/date-fns/issues/376

This is supported by Luxon via fromISO(value, {setZone: true}):

https://moment.github.io/luxon/docs/class/src/datetime.js~DateTime.html#static-method-fromISO

rjwats, 2021-01-17:
Experiments with Luxon instead of Moment yield an which is unchanged in size. Though there are some benefits to Luxon over Moment, in particular immutable DateTime instances.

Really we want a libary which supports both tree shaking and can handle retaining the offset when parsing ISO strings. It looks like this support may be coming in date-fns at some point, and a workaround might do for now:

https://github.com/date-fns/date-fns/issues/376

rjwats, 2021-01-17:
I've expeimented with all of the popular date libraries now. We use the library for formatting the dates on the NTPStatus screen and handling the manual configuration of date when NTP is disabled (there is a bug with this in master ATM).

I think the most appealing option would be to remove the dependency on a date lib altogether to reduce the UI artefact to a minimum. The change below achieves this, I have some more testing to do with different timezones/devices but it seems fine:

https://github.com/rjwats/esp8266-react/pull/221

This leaves the choice of a date library to the user extending the framework (if they decide they need one) and it reduces the bundle size by ~20k with no perceivable difference to the end user.

Here are the relative sizes for the various libraries:

- Moment (current): **187k**
- Date Fns: **175k**
- Luxon: **188k**
- DayJs: **174k**
- No Date Lib: **168K**

closed 2021-01-29 by rjwats