rjwats/esp8266-react #172: When selecting a network from the scanner hardcoded hostname is applied

rjwats, 2020-07-08:
A little change to the mount behaviour here should resolve this:

[image]

proddy, 2020-07-17:
the hostname is still hard-coded though? Could it not somehow take the value from FACTORY_WIFI_HOSTNAME? (I would also argue FACTORY_AP_SSID should be the same and there is only one hostname which is the project name). I'm only raising this point because this on my list of files I need to modify when I fetch a new copy of the library, and the list is getting bigger

rjwats, 2020-07-17:
The code snippet in the comment above shows the problem, for my reference, not a proposed solution. 

Unless I've missunderstood your comment, I think the PR addresses does address the hard coding by retaining the hostname already configured (using the existing configured hostname):

https://github.com/rjwats/esp8266-react/pull/173/files

proddy, 2020-07-17:
yup, you're absolutely right. Sorry missed that.

closed 2020-07-19 by rjwats