rjwats/esp8266-react #177: Update default esp8266 FS implementation to be LittleFS

rjwats, 2020-07-18:
Update default esp8266 FS implementation to be LittleFS

proddy, 2020-07-18:
Also keep in mind that LittleFS doesn't support directory structures natively. The filename includes the path (see https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#littlefs-file-system-limitations).  Although the code in `FactoryResetService.cpp` still seems to work:

```
  Dir configDirectory = fs->openDir(FS_CONFIG_DIRECTORY);
  while (configDirectory.next()) {
    fs->remove(configDirectory.fileName());
  }
```

later it may just be easier to make all the settings file names relative, drop the hardcoded "config/" and explicitly add the directory name in FSPersistance class. Also change it to something more obscure in case anyone also decides to use "config" in a filename.

rjwats, 2020-07-18:
It doesn't? Weird. Is that just the Arduino wrapper maintaining backwards
compatibility of FS.h because of the SPIFFS's non-POSIX semantics?

I was under the impression that LittleFS was very much more like a POSIX
filesystem (directories, atomic operations etc..)

rjwats, 2020-07-18:
It's worse, due to POSIX semantics the API now returns the file's name rather than the file's path when you call fileName(). So the factory reset feature is broken with this change. Fixed this in the branch.

At least LittleFS is an improvement over SPIFFS - POSIX behaviour makes much more sense.

Few more tests to do...

rjwats, 2020-07-18:
OK, I think the branch is good now.

I think the requirement to have the user initialize the filesystem in main.c is awkward. I never liked what I had to do to get this to work correctly with both ESP32 and ESP8266 but it was done because of the assumption you may be using an SD based file system and may want to supply that instead of a flash based FS.

[image]

Should the framework initialize the file system on the call to begin() instead?

The framework could expose the filesystem on a getFS() function, and further initializations could utalize that:

[image]

This way the user doen't need to know anything about the filesystem implmentation or the specifics of "DeviceFS".... probably better right?

rjwats, 2020-07-19:
I've made the change I suggested above in the branch... I think it's probably better to have the framework initialize the file system.

What do we think?

proddy, 2020-07-19:
I think its the right thing to do. Simplifies things.

closed 2020-07-19 by rjwats