rjwats/esp8266-react #178: MQTT Homeassistant auto discovery

tichaonax, 2020-07-19:
I followed the example for mqtt and homeassistant picks up my devices automatically. However this only happens if the topic has "light" as part of the topic. I want a Switch card. Everything with homeassistant/light/device is picked up and shows under Light card, homeassistant/switch/device does not, however both show up in MQTT. I want some devices to show under Switch card.

Am l doing something wrong.

rjwats, 2020-07-19:
Quite possibly your topics aren't configured correctly for suto
discovery... have you got a link to the code?

tichaonax, 2020-07-19:
The code can be found here. https://github.com/tichaonax/esp8266-react-scheduler
The two files with the implementation are:

esp8266-react-scheduler/src/automation/ChannelStateService.cpp
[image]
 and
esp8266-react-scheduler/src/automation/ChannelMqttSettingsService.h
[image]

The MQTT has the two entries one for light and the other the switch they differ only on the switch but only the light is picked up by homeassistant. 

See the two entries below for 85b8f.
[image]

I looked in the example and even tried putting a number infront of the switch like "switch1" as part of the topic but its not picked. What is strange is that both topics are in the MQTT but only the lights show in homeassistant.

[image]

rjwats, 2020-07-19:
It doesn't look like switches support "schema":"json" as lights do...
probably because the payload is so simple for a switch - binary value only.

rjwats, 2020-07-19:
https://www.home-assistant.io/docs/mqtt/discovery/#switches

Try poking this into your broker, it will work just fine i'm sure - adding
"schema":"json"  breaks it :)

tichaonax, 2020-07-19:
That was it, I moved the light json statement into the case block and walla! my switch showed. Thanks again!

closed 2020-07-19 by rjwats