rjwats/esp8266-react #179: MQTT support for non json payload pub/sub topics with homeassistant

tichaonax, 2020-07-20:
While trying to get Homeassistant switch play nice with the framework MQTT I noticed some deficiencies that would be nice to be part of the framework.

The current implementation assumes payload is Json data and fails to change state when publishing or subscribing to non Json payload for example the switch. We would need to extract the data from the Stateful service and customize for the switch. While you can create a configuration on the fly to tell how to sent Json data to the MQTT from Homeassistant it is not easy to tell Homeassistant to take some Json to be interpreted as ON/OFF when it just expects simple ON or OFF tag. May be we could provide a mechanism to have custom publish and subscribe methods or somehow override the default methods.

In my situation I have four channels from the same esp8266 and there are two switches and two lights. Homeassistant is now able to see the two categories, the switch works with no issues now but I had to hack the MQTT publish and subscribe framework code with some if/else and pass a flag to just get it to work but I think we need support for a more robust and elegant solution so we do not touch framework code. 

Other than that this framework is proving to be really good and useful. I am able to provide the icons I want to Homeassistant on the fly without doing anything setup in Homeassistant apart from the initial enabling discovery.

[image]

rjwats, 2020-07-20:
I agree, JsonStateReader and JsonStateUpdater are a little too constraining
for this work work nicely and there's probably a positive change to be made
there.

As a workaround you *may *be able to register the complete payload as the
on-state and off-state:

https://www.home-assistant.io/integrations/switch.mqtt/#payload_off
https://www.home-assistant.io/integrations/switch.mqtt/#payload_on

Something like this:

doc["payload_off"] = "{\"state\": \"OFF\"}";
doc["payload_on"] = "{\"state\": \"ON\"}";

Kinda horrible, and I agree the framework should be adapted, but it will
probably work for now.

rjwats, 2020-07-20:
I agree, JsonStateReader and JsonStateUpdater are a little too constraining for this work work nicely and there's probably a positive change to be made there.

As a workaround you *may *be able to register the complete payload as the on-state and off-state:

https://www.home-assistant.io/integrations/switch.mqtt/#payload_off
https://www.home-assistant.io/integrations/switch.mqtt/#payload_on

Something like this:

```
doc["payload_off"] = "{\"state\": \"OFF\"}";
doc["payload_on"] = "{\"state\": \"ON\"}";
```

Kinda horrible, and I agree the framework should be adapted, but it will probably work for now.

tichaonax, 2020-07-20:
Yes, that was the first thing l did after reading the hassio docs but that seems to work only from hassio to my device but does not work publishing from my device to hassio.

rjwats, 2020-07-20:
I tested this by poking the MQTT broker directly and it did work as
expected in both directions.

Are you sure you formatted the JSON perfectly in your payload property?
ArduinoJson uses minified formatting by default (no spaces between anything
etc...).

Maybe see what the device is sending over to make sure it matches the
configuration, this should work.

tichaonax, 2020-07-20:
Ok will try it out again. If it works l will use that approach instead so l don't touch framework codebase. Thanks.

tichaonax, 2020-07-20:
Yes very picky! it was a space that prevented updates in one direction. I reverted back the framework band aid I had put in place, thanks!

closed 2020-07-25 by rjwats