rjwats/esp8266-react #180: Question: how to implement a dynamic form

proddy, 2020-07-22:
In my project I have a web table that's dynamically built with data from coming from the ESP, similar to the way the security form is done with username/admin's and rest interface. Now I'd like to be able to make the table selectable so when a row is clicked a separate form on the same page is dynamically loaded with specific data from the ESP. Can you suggest the best way to approach this? I'm a bit stuck here.

rjwats, 2020-07-22:
I'm assuming there is a key of some description in the data for each table
row which can be used as a basis for the rest request to load/save the
nested entity.

I would pass this key as a prop into a container which can load (save if
required) the nested data and have another nested component to render the
appropriate form.

There are some good example in this thread of how to make the table work
this way: https://github.com/mui-org/material-ui/issues/4476

proddy, 2020-07-22:
Ah, I see. So refresh the whole form which contains the table and the other related information. I'll give that a go

proddy, 2020-07-26:
implemented it, thanks for the idea.

closed 2020-07-26 by proddy