rjwats/esp8266-react #181: Wifi hostname not set when using static IP

proddy, 2020-07-25:
It is still possible to set a Wifi hostname when using a static IP.  I moved the `WiFi.hostname(_state.hostname)` in `WiFiSettingsService::manageSTA()` outside the static_ip condition and it seems to work.

rjwats, 2020-07-25:
Cool thx

rjwats, 2020-07-25:
While I'm in this code... do we think it's worthwhile allowing the wifi to be turned off with an "enabled" setting?

I vaguely remember someone asking for such a feature a while ago.

proddy, 2020-07-25:
can't think of a use case where someone with an ESP wouldn't want to use wifi, unless perhaps as a bluetooth repeater node. without wifi there would be no need for a web server either

rjwats, 2020-07-25:
Originally that was my thinking too.

However a few people have since indicated that use they access point mode
to configure the device but don't connect it as a station. If the device
has no WiFi network nearby for example. You can effectively disable station
mode by leaving the ssid blank, it just can't be done explicitly at the
moment.

proddy, 2020-07-25:
oh cool, yes this a good feature. Effectively having an isolated unit not connected to the local wifi. It would mean automatically disabling the NTP, MQTT services too right?

closed 2020-09-15 by proddy