rjwats/esp8266-react #182: question: is there call via the setting services to force a 'save' of the settings?

proddy, 2020-07-26:
I wanted to be able to change the wifi username and password without immediately causing a Wifi reset, so using the `updateWithoutPropagation` function like:

```
:esp8266React.getWiFiSettingsService()->updateWithoutPropagation([&](WiFiSettings & wifiSettings) {
                                               wifiSettings.ssid = <new value here>
                                               return StateUpdateResult::CHANGED;
                                           });
```

Since none of the StatefulService registered update handlers are called I don't think the new setting values get persisted back in the FS? If correct is there a way to manually invoke a save?

rjwats, 2020-07-26:
FSPersistence class exposes the writeToFS public function which you can
call to save the current state to the file system.

proddy, 2020-07-26:
I couldn't get to the FSPersistence of the core services like Wifi, so did a work-around with
 `esp8266React.getWiFiSettingsService()->callUpdateHandlers("local"); ` which does the save.

closed 2020-07-26 by proddy

rjwats, 2020-07-26:
Yeah that's the best approach as long as you don't mind the other update
handlers also firing at the same time.